yaqwsx/PcbDraw #3: Board edge recognition issues

kwendenarmo, 2017-04-28:
I just tried this great tool, and I've found some flaws:

-   The board edge is not respected: The copper outside the board edge should be removed. For example in edge connectors.
-   With irregular board shapes, the mask is not painted correctly, (currently off the board).

The image is the copper layer (top) and the board edge along the soldermask (below). You can see these faults:

[image]

yaqwsx, 2017-04-28:
Could you send me a KiCAD board to reproduce this issue?

kwendenarmo, 2017-04-29:
Of course, you can find it in https://github.com/akornsys-rdi/USB-power-dist

yaqwsx, 2017-04-29:
Thanks, I'll try to look at it in the next few days.

yaqwsx, 2017-05-04:
PcbDraw produced invalid SVG files (integers were formatted as floats) and some editors have trouble to render them correctly. The problem is fixed and I also added support for rendering through-hole pads. Now your board looks like this:

[image]

kwendenarmo, 2017-05-04:
Awesome, great job! The "male" edge conector on the right of the board do not appear as copper (without soldermask). Is the exported image problem or does it not recognize it as a pad?

yaqwsx, 2017-05-04:
The reasons for that is that KiCAD export these two pads as an SVG path, which is for no obvious reasons properly handled. I could transform the paths to a polyline, but first I would like to spend some time for  finding the true root of problems. If I cannot solve it, I'll implement the transformation.

yaqwsx, 2017-05-04:
It should be fixed now.

kwendenarmo, 2017-05-04:
wonderful! I'll try it as soon as I can! Really great job!

closed 2017-05-04 by kwendenarmo