yaqwsx/PcbDraw #7: Allow transformations of the origin element

yaqwsx, 2017-05-29:
PCBDraw currently cannot handle any transformation of the origin element. However such a functionality is highly desirable as the library maintainers don't have to care about the SVG structure.

To implement this functionality, it is necessary to collect transformation attributes of all parent elements and build a transformation matrix out of them. Then is is possible to transform the origin point.

Any help with this issue is appreciated!

wifasoi, 2018-07-21:
I was thinking, we can use the ```<use>``` feature that svg give us. No matrix or coordinate system conversion required in the script.
Another pro is the deduplication of the symbols, you only need copy all the geometry once in a ```<def>``` block, ad use it with  ```<use xlink:href="#id" x="" y="" transform=""> (where x,y are coordinate in reference to the viewport)
For this to work the user need to define the center of the part on the 0,0(the zero of the viewport) of the svg file... (and the user don't need to mocking around with the red square)
What do you think?
I can try to implement it, but i prefer if it's cool for you to do this before i make the changes.

yaqwsx, 2018-07-22:
Implementing `use` to save image size is definitely on my ToDo list and is a highly wanted feature. PR is welcome!

About the origin - I would like to preserve the red square as there are many graphics editors treating viewports and coordinates differently. In the next few weeks I will publish an implementation which removes restriction that the origin cannot be under transformation.