yaqwsx/PcbDraw #8: Drill holes supported?

NT7S, 2017-05-29:
Are drill holes supported by PcbDraw? I'm not seeing any rendered in the drawings I generate, but I wasn't certain if that is a bug or something not yet supported.

Thank you again for the fantastic tool!

yaqwsx, 2017-05-29:
They are not supported (yet) as KiCAD plotting Python API cannot plot holes. The alternative is to walk through the components, detect holes and draw, however I haven't implemented that yet..

NT7S, 2017-05-29:
I don't know much about the KiCad API, so I appreciate the clarification.

yaqwsx, 2017-06-09:
I implemented the support for drill holes in 2d5076b:
[image]

closed 2017-06-09 by yaqwsx

NT7S, 2017-06-16:
Works like a charm here, thank you so much for this.