yaqwsx/PcbDraw #9: Fallback to 3D models?

paulo-raca, 2017-08-07:
This project is really awesome, unfortunately, the list of available models is lacking.

Maybe a long shot, but would it be possible to fallback to use the VRML 3D models already available for most parts?

E.g., there is https://github.com/tcr/vrml2svg to convert VRML into SVG (I didn't get it to work yet, but seems possible)

yaqwsx, 2017-08-15:
Hi, this could be a wanted feature, however there are much higher priorities in my life, so I am not going to implement it any time soon.

On the other hand - drawing a new model is not challenging and does not take much time. Have you considered to extend the model library of PcbDraw? ;)

paulo-raca, 2017-08-15:
Yes, I tried that and realized I suck at drawing 😖

Thanks for looking at it anyway

yaqwsx, 2017-08-15:
I am sorry for providing such an unsatisfactory response. But feel free to implement the functionality yourself and posting it as a pull request. PcbDraw is a really simple Python script.

paulo-raca, 2017-08-15:
Not unsatisfactory at all. I knew it was a long shot :)

joaoantoniocardoso, 2019-10-22:
FreeCAD can deal both VRML and SVG, maybe it can be scripted to do the job?

set-soft, 2020-07-25:
> FreeCAD can deal both with VRML and SVG, maybe it can be scripted to do the job?

It can be done, but the cost ...
The FreeCAD Python interface can save the current view as SVG with:

```
Gui.ActiveDocument.ActiveView.saveVectorGraphic("view.svg")
```

I tried it with simpl SOIC -14, it works really nice, but it generate a 297 KiB file!!

It looks nice, but is a real overkill.
The problem is that the view knows nothing about the real geometry, is just a bunch of points.

[image]

set-soft, 2020-07-25:
The following (japanese) site can convert the outline of the 3D model: https://drop-3d.com/
You need to add colors.

[image]

yaqwsx, 2020-08-05:
We could possibly use [MeshParser](https://github.com/ABI-Software/MeshParser) to load 3D models and then create a simple SVG which could serve as a template for the library artists.

However, this should be a simple separate tool, probably part of PcbDraw-Lib in my opinion. I am also worried, that we first have to solve https://github.com/yaqwsx/PcbDraw-Lib/issues/6 otherwise, there will be a mess with the libraries.

lcgamboa, 2021-02-09:
Using [stepup kicad plugin](https://github.com/easyw/kicadStepUpMod) in freecad is possible to import the kicad board and generate the outline for every selected device in top view using "3D object to 2D projection" and exporting as "flattened SVG".
The generated SVG uses paths and circles. After optimized in inkscape the file size is reduced ten times. 
The generated SVG has exactly the same scale of SVG generated by PcbDraw when open in Inkscape.

I will use this method manually to create some devices to use in PcbDraw. I am not very familiar with python and freecad, but I believe it is possible to make a python script to run on freecad to automate the creation of an outiline for each 3d model of kicad to serve as a starting point. I am also not a good designer to make an SVG by hand.

lcgamboa, 2021-02-13:
As I mentioned earlier, about the possibility of using kicad 3d models to generate an outline in SVG, I implemented a script.
I made a script to run on freecad, but as I'm not a python programmer and I'm not familiar with freecad, the code may not be of good quality, but it works.

On my computer it took two hours and a half  to convert all 3d models to more than 6200 SVG files of outilines. 

Perhaps some python programmer with knowledge of freecad will be able to improve the code. And it should also be possible to place the origin in the SVG since the StepUP workbench has support for aligning models with footprints. 

To use the script just open on freecad and run (and wait).  A version without the need to open the GUI would also be an improvement 

```
"""
dirt freecad macro pyhon script to convert all 3dmodels in svg outlines to edit and use in pcbdraw

   Copyright (c) : 2020  Luis Claudio Gambôa Lopes

   This program is free software; you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation; either version 2, or (at your option)
   any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program; if not, write to the Free Software
   Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.

   For e-mail suggestions :  lcgamboa@yahoo.com
"""
outputpath="/tmp/pcbdraw/"
model3dpath="/usr/share/kicad/modules/packages3d/"

from os import path
from os import listdir,makedirs
from os.path import isfile, isdir, join, splitext
import shutil

import ImportGui
import importSVG
           
try:
  shutil.rmtree(outputpath)
except OSError as e:
  print("Error: %s - %s." % (e.filename, e.strerror))

for dir in listdir(model3dpath):
  if isdir(join(model3dpath, dir)):   
    path_in=join(model3dpath, dir)
    path_out=join(outputpath, path.splitext(dir)[0])
    makedirs(path_out)
    for file in listdir(path_in):
       fullname_in= join(path_in, file)
       fullname_out= join(path_out, path.splitext(file)[0]+'.svg')
       if isfile(fullname_in):
         if fullname_in.endswith('.step'):
           print(fullname_in)  
           print(fullname_out)  
           doc = App.newDocument("Unnamed")
           Gui.activeDocument().activeView().viewTop()
           ImportGui.insert(fullname_in, doc.Name)
           Gui.runCommand('Std_SelectAll',0)
           Gui.runCommand('ksuTools3D2D',0)
           App.getDocument('Unnamed').removeObject('Part__Feature')
           App.getDocument('Unnamed').recompute()
           Gui.runCommand('Std_SelectAll',0)
           __objs__=[]
           __objs__.append(FreeCAD.getDocument("Unnamed").getObject("Shape2DView"))
           importSVG.export(__objs__,fullname_out)
           del __objs__
           App.closeDocument("Unnamed")
           Gui.updateGui() 
```

lcgamboa, 2021-02-14:
Project using SVG created by the [script](https://gist.github.com/lcgamboa/fd0eed10a82d0dd448677a1c1f9e18fa) : 

[image]

And using manual flat colored shapes: 
[image]

yaqwsx, 2021-05-21:
Just a small update; I just pushed a breaking change for the footprints (2bb7b343c1bb9d807c2d00104fa633390e9002a5). This change was needed for a long time as it allows for new functionality and simplifies footprint creation. You might need to slightly modify your scripts as the stupid rule "1mm = 1 user unit" no longer holds. We now use the real SVG units.

lcgamboa, 2021-05-24:
I currently use the version installed with the pip (https://pypi.org/project/PcbDraw/) which is 0.6.1, when will the new changes will generate a new version? 

My script does not change or edit the scale of the generated files. From what I saw in your script https://github.com/yaqwsx/PcbDraw-Lib/blob/master/scripts/migrate_footprint.py just add the scale in mm in the viewBox tag will fix my script?

yaqwsx, 2021-05-24:
I plan to release version v0.7 in a few weeks. I want to incorporate some internal changes.

If your script made the assumption 1 user unit = 1 mm as it used to be required, then you just have to make sure that your SVG states this. Otherwise, no changes.

yaqwsx, 2021-05-26:
I also introduced a simple tool to draw a component template for you: https://github.com/yaqwsx/PcbDraw-Lib/commit/34da350adc0de32eb8ac977941433ac1dabcfa1f

lcgamboa, 2021-05-26:
I upgraded to version 0.7.0 and my script continues to work normally without modification.
In version 0.7.0 I receive warnings on all components, using my components or the components of PcbDraw-Lib:
```
Warning: component 'LED_D5.0mm' from library 'LED_THT' has no viewBox. Cannot highlight
```
but everything works perfectly. I am using Debian 64 unstable. 

About your tool I find it very useful because it is complementary to my script, since not all kicad pcb components have an associated .stl file, but all have a .kicad_mod file.

yaqwsx, 2022-03-16:
Since version v1 (started in fead173) will be able to capture the output of KiCAD's renderer and we already have a tool for building templates, I consider this issue resolved.

closed 2022-03-16 by yaqwsx