yaqwsx/PcbDraw #10: Remap not working

DerekGn, 2017-10-17:
Nice work on this project.

I have an issue with the remap option.

I created a very basic remap file as follows:

```
{"Pin_Header_Straight_1x09_Pitch2.54mm": "Pin_Headers:Pin_Header_Straight_1x09.svg"}
```
I modified the python script slightly to dump out the dictionary contents and the path to the svg files as they are matched, see below.

The dictionary contents are:

**(u'Pin_Header_Straight_1x09_Pitch2.54mm', u'Pin_Headers:Pin_Header_Straight_1x09.svg')**

So the dictonary does contain the remap values. The get_model_file method does not seem to match the dictionary key value. For what reason I cannot determine Python is not a landguage I am that familiar with.

https://github.com/yaqwsx/PcbDraw/blob/2d5076b11aa2fa5b3a43ab58b6969a456d9696c0/pcbdraw.py#L313

I am running on debian with Python 3.5.3 Any ideas? 

```
Namespace(board='/media/sf_Shares/UsbBleDongle/UsbBleDongle.kicad_pcb', libraries=['../PcbDraw-Lib/KiCAD-base'], list_components=False, no_drillholes=False, output='gen.svg', placeholder=False, remap='remap.json', style=None)
(u'Pin_Header_Straight_1x09_Pitch2.54mm', u'Pin_Headers:Pin_Header_Straight_1x09.svg')
Please ignore following debug output of KiCAD Python API
End of KiCAD debug output
../PcbDraw-Lib/KiCAD-base/Resistors_SMD/R_1206.svg
../PcbDraw-Lib/KiCAD-base/Footprints/RN4871.svg
Warning: component 'RN4871' from library 'Footprints' was not found
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Diodes_SMD/D_0805.svg
Warning: component 'D_0805' from library 'Diodes_SMD' was not found
../PcbDraw-Lib/KiCAD-base/Diodes_SMD/D_0805.svg
Warning: component 'D_0805' from library 'Diodes_SMD' was not found
../PcbDraw-Lib/KiCAD-base/Resistors_SMD/R_0805.svg
../PcbDraw-Lib/KiCAD-base/Resistors_SMD/R_0805.svg
../PcbDraw-Lib/KiCAD-base/Resistors_SMD/R_0805.svg
../PcbDraw-Lib/KiCAD-base/Resistors_SMD/R_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Capacitors_SMD/C_0805.svg
../PcbDraw-Lib/KiCAD-base/Pin_Headers/Pin_Header_Straight_1x09_Pitch2.54mm.svg
Warning: component 'Pin_Header_Straight_1x09_Pitch2.54mm' from library 'Pin_Headers' was not found
../PcbDraw-Lib/KiCAD-base/Footprints/USB_CONN_MOLEX_48037-2200.svg
Warning: component 'USB_CONN_MOLEX_48037-2200' from library 'Footprints' was not found
../PcbDraw-Lib/KiCAD-base/Footprints/R-PDSO-N5.svg
Warning: component 'R-PDSO-N5' from library 'Footprints' was not found
../PcbDraw-Lib/KiCAD-base/Footprints/QFN-24_1EP_4x4mm_Pitch0.5mm.svg
Warning: component 'QFN-24_1EP_4x4mm_Pitch0.5mm' from library 'Footprints' was not found
```

yaqwsx, 2017-10-18:
Does the issue preserve if you omit the `.svg` suffix from the remapping file; i.e: `{"Pin_Header_Straight_1x09_Pitch2.54mm": "Pin_Headers:Pin_Header_Straight_1x09"}`

DerekGn, 2017-10-18:
Removing the svg, which I see now is incorrrect has no effect.

I modifed the get_model_file to print the lib and name. It look to me like the **if ref in remapping:** statement is not matching the ref to the dictionary key.

```
Pin_Headers
Pin_Header_Straight_1x09_Pitch2.54mm
Warning: component 'Pin_Header_Straight_1x09_Pitch2.54mm' from library 'Pin_Headers' was not found
```

I checked the remap.json file and its a UTF-8 encoded file so it should be ok as python is unicode AFAIK.

```
def get_model_file(paths, lib, name, ref, remapping):
    """ Find model file in library considering component remapping """
    for path in paths:
        if ref in remapping:
            lib, name = tuple(remapping[ref].split(":"))
        print(lib)
        print(name)
        f = os.path.join(path, lib, name + ".svg")
        if os.path.isfile(f):
            return f
    return None
```

yaqwsx, 2017-10-18:
Uh, my bad I didn't noticed one thing - the remapping is on per-component basis, not on per-footprint basis. You should remap e.g. J1 to something else. The motivation for remapping per on-component basis is to allow for differently coloured LEDs.

yaqwsx, 2017-10-25:
Is this issue solved?

DerekGn, 2017-10-25:
Yes, I will close now.

closed 2017-10-25 by DerekGn