yaqwsx/PcbDraw #12: Fixed plot origin?

Gasman2014, 2018-02-26:
Am trying to plot versions of boards that are in a VCS and then use ImageMagick to identify the differences.  Using your script, I am now able to produce files like these (where green is in the new plot but not the old and red was in the old but not the new).

[image]
diff-74b1-b1d3

[image]
diff-b1d-192a

Sometimes, however, the origin of the plot is shifted and I get this effect:

[image].

Do you know how is the plot origin is set and if it can be fixed? I have had some success by setting the <g transform="translate(-xxxx, -xx)" lines of both plots to the same values but I am unsure as to how the plotting origin is determined. I am also keen to use your xml tree to parse where the diffs have occurred.

yaqwsx, 2018-02-26:
Ok this is strange. The script itself should not move the board substrate in any way.

Could you please try save SVG returned by the get_board_substrate function? If it differs for different version of your board, then the problem is in KiCAD.

Also haven't you added a hidden reference/text/or something outside the board in PcbNew? This could lead to this behaviour.

yaqwsx, 2018-07-18:
Do you have any progress regarding this issue?

Gasman2014, 2018-07-18:
Had rather put this project on hold as I had come up with an alternative plan (but have not yet had sufficient time to implement it).

Since the error was due to an artefact off screen, I am happy for this to be closed.

yaqwsx, 2018-07-18:
Thank you for the clarification. Closing.

closed 2018-07-18 by yaqwsx