yaqwsx/PcbDraw #16: Subtract silkscreen to soldermask

wifasoi, 2018-07-17:
add an option to subtract the solder mask to the silkscreen.

yaqwsx, 2020-04-21:
What is the use case?

wifasoi, 2020-04-21:
1) This is an option on kicad when you plot gerber files.
2) You will not have text on the pads (sometimes lazy people, like me, make the silkscreen overlap with pads knowingly that the gerber plotter will subtract it for me).
3) sometimes you just want to draw an area on the silkscreen and don't bother to create the holes for the pads, so you let kicad do that for you. So when using PcbDraw to export the pcb, you'll find that some pads are missing (because they are covered by the silkscreen layer)

This is just a workaround for laziness, so I don't expect that someone (other than me) will implement it.

yaqwsx, 2020-04-21:
Makes sense and since PcbDraw tries to reproduce the actual look of PCB, this option seems like a must-have. Also, it is not that hard to implement, so I will include it when I will do the next code maintenance (probably with #17).

yaqwsx, 2020-06-30:
I thought it will be just a single line of code, unfortunately, the KiCAD's option to subtract solder mask does not work for SVG export. Therefore, I had to implement it. It should work now: c08e1796e5727f51b75045e4abdf27214a910831

closed 2020-06-30 by yaqwsx