yaqwsx/PcbDraw #18: Filter object is not subcriptable

leoheck, 2019-05-07:
Hi, I want to test your tool but I am receiving this error with filters. I don't know what to do. SDo you have any clue?

My PCB is a bord with many components. It is something unfinished. So, many components are out of the `Edge.cut` polygon.

I am using your SVG library and of course, the SVG library does not have all the components i need, but I think it is not necessary to put everything working. Am I right?

I am using `Kicad 5.1.2`, it uses Python 3, so I had to call python3 instead of python3 which is the system default. (This is why I renamed the script to `pcbdraw_lheck.py`

The version of your repo that I am using is this one (the last commit of the master branch)
[image]

```
➜ ./pcbdraw_lheck.py \
	--no-drillholes \
	~/Dropbox/kicad-plugins/PcbDraw-Lib/KiCAD-base \
	mangoh-red.kicad_pcb \
	mangoh-red.svg
Please ignore following debug output of KiCAD Python API
End of KiCAD debug output
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 642, in <module>
    board_cont.append(get_board_substrate(board, style, not args.no_drillholes, args.back))
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 380, in get_board_substrate
    process(container, f, os.path.join(tmp, svg_file), colors)
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 313, in process_board_substrate_base
    clipPath.append(get_board_polygon(extract_svg_content(read_svg_unique(source))))
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 263, in get_board_polygon
    elements.append(SvgPathItem(svg_element.attrib["d"]))
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 37, in __init__
    if path[0] != "M":
TypeError: 'filter' object is not subscriptable
make: *** [Makefile:58: svg] Error 1
```

leoheck, 2019-05-07:
Oh, this is a problem related to python 3.
I was able to fix it too.
I also found an issue with the Unicode.
I will upload my code later, I will try to convert it to something that works with python2 and 3 to make a PR.

leoheck, 2019-05-07:
It is something!
[image]

leoheck, 2019-05-07:
I got a second issue about this "highlight-offset" missing.
Is it possible to skip missing things?

I created my style.json following this here:
https://github.com/yaqwsx/PcbDraw/blob/master/doc/pcbdraw.md#writing-custom-styles
> It does not have it, so is it possible to improve that example too, adding the highlight-offset?

```
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 588, in <module>
    style = load_style(args.style)
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 547, in load_style
    .format(style_file, ", ".join(missing)))
RuntimeError: Missing following keys in style /home/lheck/Downloads/mangoh-red/svg_board_style.json: highlight-offset

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/PcbDraw/pcbdraw_lheck.py", line 593, in <module>
    print(e.message)
AttributeError: 'RuntimeError' object has no attribute 'message'
make: *** [Makefile:62: svg] Error 1
```

CornyjK, 2019-05-07:
> 
> 
> It is something!
> [image]

Just use KiCad renderer with raytracing

yaqwsx, 2019-05-08:
About skipping missing items in styles: it is intended behavior to stop on error when there is a missing value.

You can use PcbDraw with an incomplete library. The missing components simply won't be rendered and the list of such components is printed on stdout.

closed 2019-05-08 by leoheck