yaqwsx/PcbDraw #23: Solve Windows compatibility

yaqwsx, 2020-02-21:
Currently, PcbDraw has a dependency on several Python libraries. This is a problem on Windows, where KiCAD brings its own, binary incompatible, version of Python - the dependencies cannot be installed (it would be necessary to compile them using Msys2).

The proposed solution - split PcbDraw into two executables - the first one, with no dependencies, exports all necessary data from KiCAD to files, the second one (with Python dependencies) finishes the work. There will be a wrapper to automate the process; on Windows, it will run the first step using the KiCAD's Python, the second step is executed via system Python. This also solves the problem with Python2 compatibility on Windows - it should be fairly easy to write a Python2 compatible first stage.

D-Maker, 2020-07-22:
When I run the pcbdraw command of windows 10 an Error like this pops up: 
`    'Try to install:\n  ' + msg)
ImportError: MagickWand shared library not found.
You probably had not installed ImageMagick library.
Try to install:
  http://docs.wand-py.org/en/latest/guide/install.html#install-imagemagick-on-windows`

still doesn't work after installing image magick and restarting the same error appears...

yaqwsx, 2020-07-22:
I am aware of the problem, unfortunately, I am not able to solve without KiCAD migrating to Python 3 - which should happen with KiCAD 6 release.

yaqwsx, 2021-05-29:
KiKit already works on Windows with KiCAD nightly. The installation is a little clumsy, but we will provide clear instructions once stable version is released.

mikelemo, 2021-07-05:
> KiKit already works on Windows with KiCAD nightly. The installation is a little clumsy, but we will provide clear instructions once stable version is released.

Huh ok cool I'll be giving it a try soon! wonder when kicad 6 be out...

yaqwsx, 2021-07-05:
Unfortunately, the panelization still does not work as KiCAD 6 misses some essential parts of API we rely on.

yaqwsx, 2022-01-18:
With KiCAD 6 release, this is no longer relevant.

closed 2022-01-18 by yaqwsx