yaqwsx/PcbDraw #24: Wrong rotation of pad with oval shape

JarekParal, 2020-06-24:
PcbDraw generates pictures with a wrong rotation of pad with oval shape:
[image]

In Kicad both of the holes have the same direction. But:
- top has rotation = 0, x = 0.8, y = 5.5
- bottom has rotation = -90, x = 5.5, y = 0.8
[image]

Original file which were used as input for KiKit web page command:
[YellowMotorEncoder_board.zip](https://github.com/yaqwsx/PcbDraw/files/4826703/YellowMotorEncoder_board.zip)

JarekParal, 2020-06-24:
You can test fix on the YellowMotorEncoder board: https://roboticsbrno.github.io/RB0003-YellowMotorEncoder/

yaqwsx, 2020-06-27:
Fixed in f231ac2

closed 2020-06-27 by yaqwsx