yaqwsx/PcbDraw #41: Find board origin and crop out empty space

FFY00, 2020-07-10:
Hey, currently pcbdraw only crops out at the vertical bottom limit of the board. Would it be possible to do the same in the other 3 side?

Example:
[image]

set-soft, 2020-07-11:
Nice board!
Did you try with this:
https://github.com/yaqwsx/PcbDraw-Lib/blob/master/scrips/resize_canvas.sh

FFY00, 2020-07-11:
It is! I did not design it :sweat_smile:.

I haven't tried that, I will have a look tomorrow. It would still be cool to have it automatically done, would simply populate and such.

yaqwsx, 2020-07-11:
Currently, PcbDraw uses a board bounding box reported by KiCAD. This bounding box includes all items in the drawing (even the ones not used by PcbDraw). Cropping to only visible items makes sense. There are three ways to implement it:
- crawl all board items, filter only the visible ones and compute their bounding box - probably a good compromise between the complexity of implementation and dependencies,
- crawl generated SVG and crop it (probably hardest to implement, most robust),
- use Inkscape to crop it (basically what the script mentioned above does) - simplest to implement, but brings another dependency on Inkscape.

I will think about it and try to implement it. Meanwhile, feel free to use the script above.

set-soft, 2020-07-11:
> I will think about it and try to implement it. Meanwhile, feel free to use the script above.

Please try to avoid the Inkscape dependency, is 172 MiB, is ok for my desktop, but an overkill for a docker image.
I already avoided it for SVG conversions (is the default for `convert` tool).

FFY00, 2020-07-11:
I am not really aware of the API pcbnew exposes, but if you can iterate through the items and get the origin and size, it should be fairly straight forward to calculate the min and max x and y.

If this is difficult to do with the available API, you could pull inkscape as an *optional* dependency, disabling the functionality when it is missing.

yaqwsx, 2020-07-12:
Don't worry, I do not plan to use Inkscape as it is a heavy dependency.

yaqwsx, 2020-08-18:
Today I accidentally found [svgelements](https://github.com/meerk40t/svgelements). From the brief look, it looks exactly like what I need. It would also allow for dropping the `SvgPathItem` and replacing it with something more robust.

joaoantoniocardoso, 2020-08-19:
Just to mention: that inkscape script doesn't work here, I had to use:
```
inkscape -D --export-filename=out.svg --verb=FileQuit in.svg
```

And yeah, inkscape is a big dependency...

yaqwsx, 2021-05-21:
This was addressed in 27dafbc6acb31fb294cf749b3d06ee73f185ea6a

closed 2021-05-21 by yaqwsx