yaqwsx/PcbDraw #42: Live examples

obbardc, 2020-07-30:
It would be great to host the live examples on gitlab pages or similar :-)

yaqwsx, 2020-07-30:
There's no problem with the technical point of view, the problem is to get a list of projects that can be used for such live examples.

obbardc, 2020-07-30:
how about some of the projects from https://kicad-pcb.org/made-with-kicad/ ?
maybe a github action could pull the latest kicad source and build it :)

yaqwsx, 2020-07-30:
Technically, there is no problem like I said. 

Using a board from the KiCAD page requires to draw libraries, which is out of the scope of my free time - that's why I am looking for a list of projects already using PcbDraw.

yaqwsx, 2020-08-05:
Possibly, @FFY00, @wifasoi, or @set-soft have some boards that they would be willing to share for an example page. Do you?

wifasoi, 2020-08-07:
I needed to create a build guide for a old project anyway... I thin i can share that after I finish the guide.
Otherwhise i can bush up the "IPC component"  svg generator and populate some example from the https://kicad-pcb.org/made-with-kicad/ page

nobodywasishere, 2021-04-27:
Some ones it may be easy to draw components for:

https://github.com/dmitrystu/Nucleo2USB/blob/master/Nucleo2USB.kicad_pcb
<details>
<pre>
component 'AMP10118194-0001LF' from library 'CONNECTORS'
component 'DS1021-1X03S61' from library 'CONNFLY'
component 'DS1023-04-2X19A11.back' from library 'CONNFLY'
component 'KiCad-Logo_Copper_11mmx5mm' from library 'logo'
component 'LED-0603' from library 'OPTOEL'
component 'OSHW_FS_10MM' from library 'logo'
component 'SMD-0603C' from library 'SMD-CMN'
component 'SMD-0603R' from library 'SMD-CMN'
component 'SO127P050X039-8' from library 'SMD-QFN'
component 'SOD-323F' from library 'SMD-SOT'
component 'SOT-23-5' from library 'SMD-SOT'
component 'SOT-23-6' from library 'SMD-SOT'
</pre>
</details>

https://github.com/ciaa/Hardware/blob/master/PCB/NXP/ciaa-nxp.kicad_pcb
<details>
<pre>
component '1pin' from library ''
component '4SMD' from library ''
component '4SOJ.back' from library ''
component 'BK-913-ND.back' from library ''
component 'bornier2' from library ''
component 'bornier3' from library ''
component 'c_0603.back' from library ''
component 'c_0603' from library ''
component 'c_1206_tantalio_1.back' from library ''
component 'c_1206_tantalio_1' from library ''
component 'c_1206_tantalio.back' from library ''
component 'c_1206_tantalio' from library ''
component 'CASE-403-03.back' from library ''
component 'c_elec_10x10.5' from library ''
component 'c_elec_8x10' from library ''
component 'D2PACK5-LM2596.back' from library ''
component 'Diode-SMC_Standard_RevA_26July2010.back' from library ''
component 'DO214AA-1-2' from library ''
component 'FA238V-XTAL.back' from library ''
component 'FA238V-XTAL' from library ''
component 'Fiducial_1mm_BOT' from library ''
component 'Fiducial_1mm_TOP' from library ''
component 'G5V-2' from library ''
component 'GS2.back' from library ''
component 'GS2' from library ''
component 'GS3.back' from library ''
component 'GS3' from library ''
component 'K22X-E9P-NJ' from library ''
component 'LED-0603' from library ''
component 'LQFP144-M05' from library ''
component 'LQFP64-M05.back' from library ''
component 'MF-MSMF.back' from library ''
component 'MF-MSMF' from library ''
component 'MF-SM030.back' from library ''
component 'MiniMelf.back' from library ''
component 'pin_array_10x2' from library ''
component 'PIN_ARRAY_2X1' from library ''
component 'PIN_ARRAY_2X2' from library ''
component 'PIN_ARRAY_3X2' from library ''
component 'pin_array_4x2' from library ''
component 'PIN_ARRAY_4X2' from library ''
component 'Pin_Array_5x2_1.27mm_SMD' from library ''
component 'PIN_ARRAY_6X2' from library ''
component 'PUSH_SWITCH' from library ''
component 'QFN24' from library ''
component 'RJ45_SI-52003-F_PoE_ETH' from library ''
component 'SM0603.back' from library ''
[... 106 lines cut ...]
component 'SOT1534-2.back' from library 'armory-kicad'
component 'SOT1534-2' from library 'armory-kicad'
component 'SOT1999-1.back' from library 'armory-kicad'
component 'SOT1999-1' from library 'armory-kicad'
component 'TP.back' from library 'armory-kicad'
component 'TP' from library 'armory-kicad'
component 'TSX-3225.back' from library 'armory-kicad'
component 'TSX-3225' from library 'armory-kicad'
component 'UDFN-8.back' from library 'armory-kicad'
component 'UDFN-8' from library 'armory-kicad'
component 'WLCSP-6.back' from library 'armory-kicad'
component 'WLCSP-6' from library 'armory-kicad'
component 'WSON-12.back' from library 'armory-kicad'
component 'WSON-12' from library 'armory-kicad'
component 'WSON-8.back' from library 'armory-kicad'
component 'WSON-8' from library 'armory-kicad'
component 'WURTH_632712000011.back' from library 'armory-kicad'
component 'WURTH_632712000011' from library 'armory-kicad'
component 'X1SON-2.back' from library 'armory-kicad'
component 'X1SON-2' from library 'armory-kicad'
component 'X2SON-4.back' from library 'armory-kicad'
component 'X2SON-4' from library 'armory-kicad'
</pre></details>

Dealing with all the different libraries is going to be weird though (and one of them has empty libraries). I may get around to these, just wanted to keep them in one location.