yaqwsx/PcbDraw #46: `versioneer` should be a `setup_requires`

FFY00, 2020-08-05:
Looks like `versioneer` should be a `setup_requires`, right?

yaqwsx, 2020-08-05:
Oh, stupid me. Of course: c0dfd9a

Do you need a v0.6.1 release or is it OK to include the change in a later release?

FFY00, 2020-08-05:
And all in all, I think there are much better approaches, like [`setuptools_scm`](https://github.com/pypa/setuptools_scm) and using `importlib.metadata` to fetch the version.

```python
try:
    try:
        from importlib import metadata as importlib_metadata
    except ImportError:
        import importlib_metadata
    __version__ = importlib_metadata.version(__package__)
except ImportError:
    __version__ = 'unknown'
```

FFY00, 2020-08-05:
No need for a new release.

closed 2020-09-16 by yaqwsx