yaqwsx/PcbDraw #48: Problems generating PNGs

set-soft, 2020-08-25:
This is not PcbDraw fault, so take it as a warning, suggestion, whatever:
I was trying to generate PCB renders (no components) for the Spora project: [Site](https://github.com/INTI-CMNB/spora), [CI/CD example](https://gitlab.com/set-soft/kicad-ci-test-spora)
The PCB is the size of a coin, and also has circular aspect.
When generating the SVG all goes OK (well I get the extra space around, but we know it)

[image]

But when I ask for a PNG ... 

[image]

This is ImageMagick doing something wrong. I know it because I got the same using the convert command. May be this is a bug in the Debian testing package, not sure why it isn't using `librsvg`.
So I installed `librsvg2-bin`, the code I use as ImageMagick backend for conversions in my docker images, and used it directly:

```
rsvg-convert -d 600 -p 600 spora-top.svg > pp.png
convert -trim pp.png pp2.png
```
And *voilà*!

[image]

For JPG I used:

```
convert -trim pp.png -quality 85% pp2.jpg
```

[image]

I think this is a regression for you, if I'm right you switched from external commands to use ImageMagick as a module, but this seems to be more reliable (and a good way to get rid of the extra space while you solve it in the SVG ;-)

yaqwsx, 2020-08-25:
Thanks for reporting this. Conversion of the SVG into bitmap is known to be tricky. Let me give you a brief overview of my experiences/history:

I didn't switch from calling external programs to ImageMagic. I was using cairo & librsvg directly. It wasn't a good choice as they are troublesome to install on some systems and also, I was able to find boards on which the conversion failed terribly. Therefore, I switched to Wand, which uses ImageMagick.

ImageMagic uses various backends to perform the conversion -  one of them is built-in with no external dependencies (which sometimes fail - just like in the case of your board), the other one uses librsvg and the last one uses Inkscape as backend. The only reliable solution in my experience is to use Inkscape as a backend. If Inkscape is installed on your system, ImageMagick will use it. 

The current solution works the best so far (if Inkscape is installed) and is also the easiest to maintain. Therefore, I do not plan to switch to calling `rsvg-convert` directly and I will rather add a note in the readme about Inkscape backend.

If this workflow works better for you, I suggest you write a wrapper script for PcbDraw which will call PcbDraw, outputs temporary SVG, which you then convert to a bitmap with possible trimming.

set-soft, 2020-08-25:
> If this workflow works better for you, I suggest you write a wrapper script for PcbDraw which will call PcbDraw, outputs temporary SVG, which you then convert to a bitmap with possible trimming.

I'm already doing it.
 
> BTW: Have you considered to use https://github.com/yaqwsx/KiKit in your CI for generating fabrication files?

Is in my TODO list, but I have other things to do before it.

BTW: I have Inkscape and librsvg installed and ImageMagick isn't using them.

yaqwsx, 2020-08-31:
Is there a way to get your environment (e.g. Docker container) which features this behavior so I can debug what's wrong and why is Inkscape not used for the conversion?

yaqwsx, 2021-04-18:
Hi, I just run into a similar problem with a new machine. The problem lies in ImageMagick and its (in)compatibility with various delegates. You can find out what exactly is wrong with:

```
pcbdraw <option> board output.svg
convert -verbose output.svg output.png
```

It will give you a list of commands that were tried in order to convert the image. With that information, it is possible to detect what went wrong and how you need to reconfigure ImageMagick.

yaqwsx, 2021-05-21:
More details can be find in https://github.com/ImageMagick/ImageMagick/issues/3564

yaqwsx, 2021-05-29:
Ok, an update on the problem; I discovered that in some places we first use a mask defined later. This breaks e.g., lib-rsvg. This was fixed in https://github.com/yaqwsx/PcbDraw/commit/c8867fdd837fb0b05086518c89f1351e517f6057

However, when ImageMagick decides to use MVG based converter, things break terribly as it does not support masks. In theory, we could bake the masks in, but it seems like a lot of work not worth the effort.

yaqwsx, 2022-03-04:
I decided to opt-out out of ImageMagick and handle the conversion directly via available tools. This should, hopefully, prevent any troubles with misconfigured ImageMagick. The changes are available in the upstream version.

closed 2022-03-16 by yaqwsx