yaqwsx/PcbDraw #52: [MacOS] ModuleNotFoundError: No module named 'pcbnew'

martijnvdwoude, 2020-09-16:
```
Traceback (most recent call last):
  File "/usr/local/bin/pcbdraw", line 5, in <module>
    from pcbdraw.pcbdraw import main
  File "/usr/local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 18, in <module>
    import pcbnew
ModuleNotFoundError: No module named 'pcbnew'
```

Anyone running PcbDraw on MacOS successfully? I'm not this familiar with Python and I can't seem to find any related issues for MacOS. I installed through homebrew.

```
Application: Pcbnew
Version: (5.1.6-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.39.2
Platform: macOS Catalina Version 10.15.6 (Build 19G73), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.72.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

set-soft, 2020-09-16:
I know nothing about MacOSX, but you should look for `pcbnew.py`.

From my notes: 
- KiCad configuration should be in ~/Library/Preferences/kicad/
- Most files should be in /Library/Application Support/kicad/ or in ~/Library/Application Support/kicad/

Try defining the environment variable `PYTHONPATH` so it includes the path to pcbnew.py

martijnvdwoude, 2020-09-16:
Thank you for the quick nudge in the right direction! `export PYTHONPATH=/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages` got me a step further, though I'm now getting:

```
Traceback (most recent call last):
  File "/usr/local/bin/pcbdraw", line 5, in <module>
    from pcbdraw.pcbdraw import main
  File "/usr/local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 18, in <module>
    import pcbnew
  File "/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.py", line 15, in <module>
    import _pcbnew
ImportError: dynamic module does not define module export function (PyInit__pcbnew)
```

The more I dig up about python support for KiCAD on MacOS the more clear it becomes that this might not be as useable as one would hope. But still keeping my fingers crossed that someone here has a working setup, if so it would be appreciated if you'd let me know!

yaqwsx, 2020-09-16:
It has the same problem as KiKit: https://github.com/yaqwsx/KiKit/issues/21 Until the next release of KiCAD there's not much I can do. 

The workaround is to use Docker image yaqwsx/kikit which contains PcbDraw.

martijnvdwoude, 2020-09-16:
Thank you for pointing me to that container, it actually works now and generates images. I'm getting component warnings for footprints and vias are also not rendering yet but I'll have to play around with it. Thanks!

yaqwsx, 2021-07-30:
Just a small update; with KiCAD nightly (v5.99) PcbDraw should work out of the box on MacOS.

closed 2022-01-18 by yaqwsx