yaqwsx/PcbDraw #55: Problem trying to use populate.

GonzaELP, 2020-10-10:
Hi!, 
I'm having some trouble trying to populate a board, when i run populate i get the following error:

```
(kicadenv) gonzalo@ubuntu:~/.virtualenvs/kicadenv/pcb_epi_tens$ populate fuente test
Warning: extra parameter 'template'
PcbDraw failed with code 1 and output:
b'Traceback (most recent call last):\n  File "/home/gonzalo/.virtualenvs/kicadenv/lib/python3.8/site-packages/pcbdraw/pc
bdraw.py", line 21, in <module>\n    from pcbdraw import __version__\n  File "/home/gonzalo/.virtualenvs/kicadenv/lib/py
thon3.8/site-packages/pcbdraw/pcbdraw.py", line 21, in <module>\n    from pcbdraw import __version__\nImportError: canno
t import name \'__version__\' from partially initialized module \'pcbdraw\' (most likely due to a circular import) (/hom
e/gonzalo/.virtualenvs/kicadenv/lib/python3.8/site-packages/pcbdraw/pcbdraw.py)\n'

```
I've used the example source_md.md file (renamed as "fuente") to generate the population, what i did was changing the route of the .kicad_pcb file, this is the file content:

```
---
img_name: img/populating_{}.png
type: md
board: PlacaEPI_TENS.kicad_pcb
libs: kicad-default
...

# Demo population manual

Lorem ipsum dolor sit amet, consectetuer adipiscing elit. Itaque earum rerum hic tenetur a sapiente delectus, ut aut reiciendis voluptatibus maiores alias consequatur aut perferendis doloribus asperiores repellat. Vestibulum fermentum tortor id mi. Nulla turpis magna, cursus sit amet, suscipit a, interdum id, felis.

- [[front | ]] This is the front side of the board we are populating
- [[back | ]] This is the back side of the board we are populating
- [[front | RV1, RV2 ]] First, populate RV1 and RV2. Basically, any description could be here.
- [[front | U2 ]] Let's populate U2!

You can put a paragraph of text between the population steps.

- [[back | R24 ]] We can also populate a component on the other side

And this is the end of the demo.
```

I've also generated the .svg files of the board using pcbdraw without any major issues, i've only got some warnings related to some missing libraries:

```
Warning: component 'R_0805_HandSoldering' for footprint 'R7' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R6' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R5' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R4' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R3' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R2' from library 'EMorales' was not found
Warning: component 'R_0805_HandSoldering' for footprint 'R1' from library 'EMorales' was not found
```
Since i didn't need to plot the footprints or  component's images, I´ve ignored these warnings. Could these warnings and the error I had when I tried to populate the board be related?. 

Thank you for the great job you're doing with this project!
Gonzalo.-

yaqwsx, 2020-10-10:
How did you install PcbDraw? It seems that it is installed incorrectly. The correct way is to use pip to get the latest version (possibly in venv).

GonzaELP, 2020-10-10:
Hi Jan, thanks for the quick answer.
I installed PcbDraw in a virtual env using pip. According to _pip list_ the PcbDraw version i have is : **0.6.0**.
The OS i'm using is:  Ubuntu server 20.04.1 LTS
And the Python version: Python 3.8.5

yaqwsx, 2020-10-11:
Oh, you are affected by problem fixed in 0113a35ebe3811bad2a4768aa1a7125d9b0a2d81. No release was made since then. So I released version 0.6.1 containing this bug fix. Please upgrade Pcbdraw and try it again.

GonzaELP, 2020-10-11:
Thank you so much for your help, problem solved!, Now I can use populate to generate the PCB (just the board, no components).

closed 2020-10-12 by yaqwsx