yaqwsx/PcbDraw #56: Draw only components of the current side of board

Salamandar, 2020-10-22:
I'm trying to draw a PCB with PCBDraw, but i'm facing an issue:

* When I draw the front side of the board, it also draws the components present on the BACK of the board.
* Same when I draw the back.

I could write 2 filtering files, but that's a bit long. Would it be possible to filter every SMD components from the other side of the board ?
Thanks.

yaqwsx, 2020-10-22:
Could you provide me with an example board and command? This should work.

Salamandar, 2020-10-26:
Here is the kicad_pcb (Github doesn't like this file type so it's zipped), the corresponding image and the command: 

[theyseemerolling-elec.kicad_pcb.zip](https://github.com/yaqwsx/PcbDraw/files/5438242/theyseemerolling-elec.kicad_pcb.zip)

[image]

```
pcbdraw theyseemerolling-elec.kicad_pcb theyseemerolling-elec.png --remap remapping.json --placeholder
```

`remapping.json` : (without it no component is shown)
```{
    "C1":   "Capacitors_SMD:C_0805_2012Metric",
    "C2":   "Capacitors_SMD:C_0805_2012Metric",
    "C3":   "Capacitors_SMD:C_0805_2012Metric",
    "C4":   "Capacitors_SMD:C_0805_2012Metric",
    "C5":   "Capacitors_SMD:C_0805_2012Metric",
    "C6":   "Capacitors_SMD:C_0805_2012Metric",
    "C7":   "Capacitors_SMD:C_0805_2012Metric",
    "C8":   "Capacitors_SMD:C_0805_2012Metric",
    "R1":   "Resistors_SMD:R_0805_2012Metric",
    "R2":   "Resistors_SMD:R_0805_2012Metric",
    "R3":   "Resistors_SMD:R_0805_2012Metric",
    "R4":   "Resistors_SMD:R_0805_2012Metric",
    "R5":   "Resistors_SMD:R_0805_2012Metric",
    "R6":   "Resistors_SMD:R_0805_2012Metric",
    "R7":   "Resistors_SMD:R_0805_2012Metric",
    "R8":   "Resistors_SMD:R_0805_2012Metric",
    "R9":   "Resistors_SMD:R_0805_2012Metric",
    "R11":  "Resistors_SMD:R_0805_2012Metric",
    "R13":  "Resistors_SMD:R_0805_2012Metric",
    "R15":  "Resistors_SMD:R_0805_2012Metric",
    "R16":  "Resistors_SMD:R_0805_2012Metric"
}
```

yaqwsx, 2020-10-26:
Oh, you didn't specify in the initial description that you are interested in placeholders. The placeholders are intended to be displayed on both sides - when you have a THT component it also has a backside. That's why placeholders are showed.

What is your use case for placeholders? Placeholders were originally designed as an aid for library creators to verify component placement. If you have a valid use case for them, I will consider adding a switch for only showing them on the current side. Otherwise, I don't see a problem that should be fixed.

Salamandar, 2020-10-26:
The issue is not about placeholders. I guessed they were shown for both faces and that's okay.

For example, on the left of the image, between R3 and C10, there's a capacitor that should be on the other side of the pcb. A bit lower, there's R4 and a capacitor that should not be here. 
On the right, there's the silkscreen "Motor2" and 2 components U3 and U4. The resistors **inside** those 2 components are on the back of the pcb.
Etc.

yaqwsx, 2020-10-26:
Ok, now I see it. I am sorry, I was confused by the placeholders. I'll check it out and let you know in a few days.

yaqwsx, 2020-10-27:
There was a bug in rendering the backside of the component when remapping was applied. This was fixed in a6a1f5d. Please test it. If you don't find any problems, I will make a release.

Salamandar, 2020-10-28:
Thanks, I'm gonna test it right away !!

closed 2020-10-30 by Salamandar