yaqwsx/PcbDraw #59: Add option to hide references and values

mluessi, 2020-12-18:
I have been playing around with PcbDraw and it's very neat :). One thing I'm wondering is if there is a way to not show the references and values on the silk screen layer. 

I was looking through the code to add an option for this but couldn't find where they are added to the SVG. If you can point me in the right direction, I can try to add an option to hide them.

yaqwsx, 2020-12-18:
The whole silkscreen is rendered internally in KiCAD. If your board has visible references, then they will be included in PcbDraw. If you want to batch hide references, look at [KiKit](https://github.com/yaqwsx/KiKit) - you can use command `kikit modify references --hide -p '.*' <yourboard>` to hide them.

Currently, I do not plan to add such an option to PcbDraw.

mluessi, 2020-12-18:
Thanks! I will try the KitKit workaround

closed 2020-12-19 by yaqwsx

set-soft, 2020-12-21:
Hi @mluessi !
If you just need to remove the whole "Silk" layer may be the following is enough: INTI-CMNB/PcbDraw#11
What do you think @yaqwsx ?

yaqwsx, 2020-12-21:
I think @mluessi wants to preserve silkscreen, but he only wants to only hide references.

INTI-CMNB#11 seems pretty simple, so I would be willing to merge. I am just worried about having a large number of "single-use" options. Does not-rendering silkscreen have a proper use case? Wouldn't it be better if just allow the users to specify a general plot plan from a file and possibly prepare some of them (e.g., full-stack, no-silk)? Note that I have no answers for this, I am just thinking loudly. I welcome any other opinions.