yaqwsx/PcbDraw #62: problem about absolute position

scarrrr316, 2021-01-17:
when I using pcbdraw, it told me that paths with absolute position are not supported ,how can i fix it 
↓
[image]

You seem to have fixed it ? https://github.com/yaqwsx/PcbDraw/commit/9a407205fc86150246ea1c28225341d0c87c5d2a

why still show this error. thxxxxx

yaqwsx, 2021-01-17:
Your report contains too little information to actually help you - e.g., having the input KiCAD board would be helpful.

It is possible, that 9a40720 fixes that. You can install the latest upstream version of PcbDraw by:
```
pip3 install git+https://github.com/yaqwsx/PcbDraw@master
```

scarrrr316, 2021-01-17:
I installed the latest upstream version of PcbDraw

my file :
[lixie-.zip](https://github.com/yaqwsx/PcbDraw/files/5826537/lixie-.zip)

preview:
[image]

```
scarrrr@DESKTOP-B4R53N8:~/.local/share$ pcbdraw /mnt/c/pcbStencilGenerator/lixie+.kicad_pcb /mnt/c/pcbStencilGenerator/a
bc.PNG

Traceback (most recent call last):
  File "/home/scarrrr/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(main())
  File "/home/scarrrr/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 793, in main
    board_cont.append(get_board_substrate(board, style, not args.no_drillholes, args.back))
  File "/home/scarrrr/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 445, in get_board_substrate
    process(container, f, os.path.join(tmp, svg_file), colors, boardsize)
  File "/home/scarrrr/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 329, in process_board_substrate_base
    clipPath.append(get_board_polygon(extract_svg_content(read_svg_unique(source))))
  File "/home/scarrrr/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 277, in get_board_polygon
    elements.append(SvgPathItem(svg_element.attrib["d"]))
  File "/home/scarrrr/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 52, in __init__
    raise SyntaxError("Only paths with absolute position are supported")
SyntaxError: Only paths with absolute position are supported
```

thx !!!!!!!!!!

yaqwsx, 2021-01-17:
I just tried your board with the upstream version; it seems to work just fine: 
[image]

Could you please provide me with the output of `pcbdraw --version`?

scarrrr316, 2021-01-18:
> I just tried your board with the upstream version; it seems to work just fine:
> [image]
> 
> Could you please provide me with the output of `pcbdraw --version`?

```
scarrrr@DESKTOP-B4R53N8:/mnt/c/Users/zysee$ pcbdraw --version
PcbDraw 0.6.1
```

I just installed it yesterday following the installation guide

yaqwsx, 2021-01-18:
This is not the upstream version; it should be something like `PcbDraw 0.6.1+6.g83a2a25`. Try to specify `--force` option for the installation command I posted above.

scarrrr316, 2021-01-18:
> This is not the upstream version; it should be something like `PcbDraw 0.6.1+6.g83a2a25`. Try to specify `--force` option for the installation command I posted above.

ohhhhh , now is PcbDraw 0.6.1+5.g4e8f08b version , how can i install the pcbdraw lib ? 
[image]
like this ? ↑

```
scarrrr@DESKTOP-B4R53N8:/mnt/c/Users/zysee$ pcbdraw --version
PcbDraw 0.6.1+5.g4e8f08b
scarrrr@DESKTOP-B4R53N8:/mnt/c/Users/zysee$ pcbdraw /mnt/c/pcbStencilGenerator/lixie-.kicad_pcb /mnt/c/pcbStencilGenerator/abc.PNG
Warning: component '1X03' for footprint 'JP1' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED1' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED2' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED3' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED4' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED5' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED6' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED7' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED8' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED9' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED10' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED11' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED12' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED13' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED14' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED15' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED16' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED17' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED18' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED19' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED20' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED21' from library 'easyeda' was not found
Warning: component 'WS2812B-NARROW' for footprint 'LED22' from library 'easyeda' was not found
Warning: component '1X03' for footprint 'JP2' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C1' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C2' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C3' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C4' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C6' from library 'easyeda' was not found
Warning: component 'C0603' for footprint 'C7' from library 'easyeda' was not found
Warning: component 'MountingHole_3.00mm' for footprint '' from library 'AutoGenerated' was not found
Warning: component 'MountingHole_3.00mm' for footprint '' from library 'AutoGenerated' was not found
Warning: component 'MountingHole_3.00mm' for footprint '' from library 'AutoGenerated' was not found
Warning: component 'MountingHole_3.00mm' for footprint '' from library 'AutoGenerated' was not found
```

thxxxxxxx

scarrrr316, 2021-01-18:
> 我只是在上游版本上尝试过您的主板；似乎工作正常：
> [image]
> 
> 您能否提供我的输出`pcbdraw --version`？

> I just tried your board with the upstream version; it seems to work just fine:
> [image]
> 
> Could you please provide me with the output of `pcbdraw --version`?

can you also provide the command that you used? thxx

yaqwsx, 2021-01-18:
PcbDrawLib is already pre-packed with PcbDraw. The library just does not contains components used in your board - you use a non-standard library `easyeda`. You have to draw one yourself. 

I used the exact same command as you in your screenshots. 

PS: Could you please reply via GitHub web instead of e-mail replies? The replies include a lot of extra text thus are harder to follow.

scarrrr316, 2021-01-18:
> PcbDrawLib is already pre-packed with PcbDraw. The library just does not contains components used in your board - you use a non-standard library `easyeda`. You have to draw one yourself.
>
> I used the [...]

huh ？？i am using github web to reply ，i didnt use email    [image]
[image]

ok ，i will draw it ，if the library does not contains  those components ,pcbdraw still generate image of board , right?
but it generated a whole black image 
[image]

how can i fix it

scarrrr316, 2021-01-18:
[image]
no problem with svg ,but jpg and png

yaqwsx, 2021-01-19:
It seems that the Wand library used for conversion from SVG to PNG cannot perform its job. This could be related to #48. Do you have all the dependencies (e.g., Inkscape) installed in your environment where you run PcbDraw?

scarrrr316, 2021-01-19:
I installed both `inkscape` and  `librsvg2-bin` but still generate empty image using `pcbdraw /mnt/c/PRISM/PRISM.kicad_pcb /mnt/c/PRISM/PRISM.png` command

also, I cant install wand using ` sudo apt-get install python-wand ` so i installed with `sudo apt-get install libmagickwand-dev `
```
scarrrr@DESKTOP-B4R53N8:~$ sudo apt-get install python-wand
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package python-wand is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-wand' has no installation candidate
```

i successfully converted svg to png using `rsvg-convert -d 600 -p 600 /mnt/c/PRISM/PRISM.svg > /mnt/c/PRISM/prism.png`

[image]

yaqwsx, 2021-05-21:
Was the problem solved?

yaqwsx, 2021-05-29:
As there is no activity for more than a week, I consider the issue resolved.

closed 2021-05-29 by yaqwsx