yaqwsx/PcbDraw #63: Fails to draw boards with circular edges

elektroniikka, 2021-02-26:
I was just trying out this cool tool and it works perfectly fine with boards that have straight edges. But when the board edge is curved, it crashes.

Here's the sample board I tried,
[image]

and here's the crash log
```
Traceback (most recent call last):
  File "/home/marcus/.local/bin/pcbdraw", line 11, in <module>
    sys.exit(main())
  File "/home/marcus/.local/lib/python3.6/site-packages/pcbdraw/pcbdraw.py", line 793, in main
    board_cont.append(get_board_substrate(board, style, not args.no_drillholes, args.back))
  File "/home/marcus/.local/lib/python3.6/site-packages/pcbdraw/pcbdraw.py", line 445, in get_board_substrate
    process(container, f, os.path.join(tmp, svg_file), colors, boardsize)
  File "/home/marcus/.local/lib/python3.6/site-packages/pcbdraw/pcbdraw.py", line 329, in process_board_substrate_base
    clipPath.append(get_board_polygon(extract_svg_content(read_svg_unique(source))))
  File "/home/marcus/.local/lib/python3.6/site-packages/pcbdraw/pcbdraw.py", line 284, in get_board_polygon
    outline = [elements[0]]
IndexError: list index out of range
```

yaqwsx, 2021-02-26:
Could you share the '*.kicad_pcb' file, so I can easily reproduce the issue?

elektroniikka, 2021-02-26:
Here you go :) [PCBDrawTest.zip](https://github.com/yaqwsx/PcbDraw/files/6049367/PCBDrawTest.zip)

The command I tried was 
```
$ pcbdraw PCBDrawTest.kicad_pcb PCBDrawTest.png
```

markasoftware, 2021-05-02:
I can confirm this issue.

yaqwsx, 2021-05-03:
Fixed in be3fbc8e464b4b220a22b3192be30b729eb08583.

closed 2021-05-03 by yaqwsx