yaqwsx/PcbDraw #66: Remapping pcb footprint <-> pcbdraw footprints instead of reference <-> pcbdraw footprint

Salamandar, 2021-05-21:
For now I'm forced to list every footprint on the pcb (30 resistors, 30 capacitors) with a remapping for each. That's a bit error-prone.

What I have in mind is something like : 
```
{
    "Resistor_SMD:R_0805_2012Metric_Pad1.15x1.40mm_HandSolder": "Resistors_SMD:R_0805_2012Metric",
    "Capacitor_SMD:C_0805_2012Metric_Pad1.15x1.40mm_HandSolder": "Capacitors_SMD:C_0805_2012Metric"
}
```

That could be named as a "compatibility table" between PCBnew lib and PCBDraw lib. That way, a remapping can still be used.

yaqwsx, 2021-05-26:
The original idea for remapping is to e.g., allow for different LED colors. What you are doing is rather hacking the feature - what you probably want is to create a symlink between the library to support the new naming. Or create a new library that reflects the naming changes in the KiCAD library. 

When the library is fixed, I don't see much of a point in remapping footprints. Instead I am thinking about introducing more general remapping for the most complicated cases: the user supplies a Python function that receives KiCAD footprint (with all text fields and other properties) and the function remaps the footprint. In that ways, we could, e.g., change the color of LEDs automatically.

Salamandar, 2021-05-26:
> What you are doing is rather hacking the feature - what you probably want is to create a symlink between the library to support the new naming.

Exactly.

Also, users may use custom components but would still want to use pcbdraw-lib without having to rename a bunch of files.

yaqwsx, 2022-03-16:
For v1 I decided to preserve the symlinks, but, create a basic `PcbDraw` library that will be the source of the truth. All other libraries will just symlink these models. The code for that was introduced in fead173 and the library will be migrated in the near future.