yaqwsx/PcbDraw #68: Shared Object Import Error

nitko12, 2021-09-01:
Hi,

I've been trying to use your code on files made in nightly v5.99 20210126 release.

I'm running it in the docker container using https://github.com/INTI-CMNB/kicad_auto nightly image which I've manually updated to latest Kicad version (default is a year old release) but I keep getting these errors:

```root@258b682816b4:/TEST_PCB_fiducial# PYTHONPATH=/usr/lib/kicad-nightly/lib/python3/dist-packages pcbdraw TEST_PCB_fiducial.kicad_pcb out.jpg
Traceback (most recent call last):
  File "/usr/bin/pcbdraw", line 11, in <module>
    load_entry_point('PcbDraw==0.6.0', 'console_scripts', 'pcbdraw')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 474, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2846, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3/dist-packages/pcbdraw/pcbdraw.py", line 18, in <module>
    import pcbnew
  File "/usr/lib/kicad-nightly/lib/python3/dist-packages/pcbnew.py", line 38, in <module>
    import _pcbnew
ImportError: libkicad_3dsg.so.2.0.0: cannot open shared object file: No such file or directory```

Any help would be appreciated as I've ran out of ideas for a fix :)

The file in questions exsists on the system here:
`./usr/lib/kicad-nightly/lib/x86_64-linux-gnu/libkicad_3dsg.so.2.0.0`
but Python can't find it for some reason.

yaqwsx, 2021-09-01:
Hi, this seems like a problem in KiCAD itself - it cannot find an internal library. I have tested PcbDraw with KiCAD v5.99 about a month ago and it was working fine.

nitko12, 2021-09-01:
Hi, after some more work I've made it do export something, but now can't get any footprints working (it just renders black)
I'll keep trying, let you know if (or when) I give up :)

Thanks for the quick reply.

closed 2021-09-01 by nitko12

yaqwsx, 2021-09-01:
Well, if the issue is not resolved, you should leave it open.