yaqwsx/PcbDraw #72: Crash error

holmesrichards, 2021-11-17:
Using PcbDraw 0.8.0 with Linux Mint 20.2 Cinnamon I get a crash:

```
Traceback (most recent call last):
  File "/home/rsholmes/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(main())
  File "/home/rsholmes/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 888, in main
    svg_to_bitmap(tmp_f.name, args.output, dpi=args.dpi)
  File "/home/rsholmes/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 629, in svg_to_bitmap
    image.read(filename=infile, resolution=dpi)
  File "/home/rsholmes/.local/lib/python3.8/site-packages/wand/image.py", line 9815, in read
    self.raise_exception()
  File "/home/rsholmes/.local/lib/python3.8/site-packages/wand/resource.py", line 222, in raise_exception
    raise e
wand.exceptions.FileOpenError: unable to open file `/tmp/magick-71547THq8ivVfSVJh': No such file or directory @ error/constitute.c/ReadImage/599
```

yaqwsx, 2021-11-17:
Please, verify first that you can convert SVG images produced by PcbDraw to PNGs via ImageMagic command-line interface. This error comes from the library Wand which binds to ImageMagic. I suspect there will be a problem with your local installation - that could be confirmed via manual conversion.

holmesrichards, 2021-11-17:
Solution was to install:

    sudo apt-get install librsvg2-bin

This maybe could be mentioned as a dependency.

closed 2021-11-17 by holmesrichards