yaqwsx/PcbDraw #79: How to install on windows 10 and kicad version 6.99 ?

Witawat, 2021-12-23:
How to install on windows 10 and kicad version 6.99 ?

i use 

C:\Program Files\KiCad\6.0\bin>python.exe -m pip install PcbDraw
Collecting PcbDraw
  Using cached PcbDraw-0.8.0-py3-none-any.whl (696 kB)
Collecting lxml
  Using cached lxml-4.7.1-cp39-cp39-win_amd64.whl (3.7 MB)
Collecting pcbnewTransition
  Using cached pcbnewTransition-0.2.0-py3-none-any.whl (4.3 kB)
Collecting mistune
  Using cached mistune-2.0.0-py2.py3-none-any.whl (24 kB)
Collecting pyyaml
  Using cached PyYAML-6.0-cp39-cp39-win_amd64.whl (151 kB)
Collecting wand
  Using cached Wand-0.6.7-py2.py3-none-any.whl (139 kB)
Requirement already satisfied: numpy in c:\program files\kicad\6.0\bin\lib\site-packages (from PcbDraw) (1.21.5)
Collecting svgpathtools
  Downloading svgpathtools-1.4.4-py2.py3-none-any.whl (66 kB)
     |████████████████████████████████| 66 kB 929 kB/s
Collecting click
  Downloading click-8.0.3-py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 2.2 MB/s
Collecting pybars3
  Using cached pybars3-0.9.7.tar.gz (29 kB)
  Preparing metadata (setup.py) ... done
Collecting colorama
  Using cached colorama-0.4.4-py2.py3-none-any.whl (16 kB)
Collecting PyMeta3>=0.5.1
  Using cached PyMeta3-0.5.1.tar.gz (29 kB)
  Preparing metadata (setup.py) ... done
Collecting scipy
  Downloading scipy-1.7.3-cp39-cp39-win_amd64.whl (34.3 MB)
     |████████████████████████████████| 34.3 MB 6.4 MB/s
Collecting svgwrite
  Downloading svgwrite-1.4.1-py3-none-any.whl (66 kB)
     |████████████████████████████████| 66 kB 2.6 MB/s
Building wheels for collected packages: pybars3, PyMeta3
  Building wheel for pybars3 (setup.py) ... done
  Created wheel for pybars3: filename=pybars3-0.9.7-py3-none-any.whl size=14093 sha256=e093c67e743f47450ce03c11e51d1fc39d614c7886ea18be4e72cc0b01b6b322
  Stored in directory: c:\users\xsoftz-pc\appdata\local\pip\cache\wheels\2c\63\6e\2fdd0384087778f0903c4dd80c8c0ee5355c551827428d7333
  Building wheel for PyMeta3 (setup.py) ... done
  Created wheel for PyMeta3: filename=PyMeta3-0.5.1-py3-none-any.whl size=16449 sha256=438891da1c672fe2a94a770624f33ebf6d3631c716ad34045daf61aafbb59ccb
  Stored in directory: c:\users\xsoftz-pc\appdata\local\pip\cache\wheels\53\58\3f\4eadb07bcff9847154678b3cb722556834b477fb4b1cf1c2d0
Successfully built pybars3 PyMeta3
Installing collected packages: svgwrite, scipy, PyMeta3, colorama, wand, svgpathtools, pyyaml, pybars3, pcbnewTransition, mistune, lxml, click, PcbDraw
  WARNING: The scripts pcbdraw.exe and populate.exe are installed in 'C:\Program Files\KiCad\6.0\bin\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed PcbDraw-0.8.0 PyMeta3-0.5.1 click-8.0.3 colorama-0.4.4 lxml-4.7.1 mistune-2.0.0 pcbnewTransition-0.2.0 pybars3-0.9.7 pyyaml-6.0 scipy-1.7.3 svgpathtools-1.4.4 svgwrite-1.4.1 wand-0.6.7

and run

C:\Program Files\KiCad\6.0\bin\Scripts>pcbdraw.exe
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\KiCad\6.0\bin\Scripts\pcbdraw.exe\__main__.py", line 4, in <module>
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pcbdraw\pcbdraw.py", line 20, in <module>
    from pcbnewTransition import pcbnew, pcbnewVersion, isV6
ImportError: cannot import name 'pcbnewVersion' from 'pcbnewTransition' (C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pcbnewTransition\__init__.py)

not work

yaqwsx, 2021-12-23:
Please, update to the latest nightly of KiCAD. This issue was addressed about 1 or 2 days ago in the Windows packaging.

yaqwsx, 2021-12-23:
I also updated the installation instruction for Windows in e575eef85c172ca60edf374f9b83415760915a70. Plese see https://github.com/yaqwsx/PcbDraw/blob/master/doc/installation.md

closed 2022-01-18 by yaqwsx