yaqwsx/PcbDraw #82: Problems using `--shrink` on circular PCB

set-soft, 2021-12-28:
I tried `--shrink` on the following PCB: [spora](https://github.com/INTI-CMNB/kicad-ci-test-spora/blob/master/pcb_main/spora.kicad_pcb) using:

```
pcbdraw --shrink 0 spora.kicad_pcb 2.png
```

And it crashed:

```
Traceback (most recent call last):
  File "/usr/bin/pcbdraw", line 33, in <module>
    sys.exit(load_entry_point('PcbDraw==0.9.0', 'console_scripts', 'pcbdraw')())
  File "/usr/lib/python3/dist-packages/pcbdraw/pcbdraw.py", line 988, in main
    postprocess_svg(tmp_f.name, args.shrink)
  File "/usr/lib/python3/dist-packages/pcbdraw/pcbdraw.py", line 824, in postprocess_svg
    shrink_svg(svgfilepath, shrinkBorder)
  File "/usr/lib/python3/dist-packages/pcbdraw/pcbdraw.py", line 778, in shrink_svg
    paths = document.paths()
  File "/usr/lib/python3/dist-packages/svgpathtools/document.py", line 261, in paths
    return flattened_paths(self.tree.getroot(), group_filter,
  File "/usr/lib/python3/dist-packages/svgpathtools/document.py", line 143, in flattened_paths
    path = transform(parse_path(converter(path_elem)), path_tf)
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 309, in transform
    return transform_segments_together(curve, transformation)
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 191, in transform_segments_together
    transformed_segs = [transformation(seg) for seg in path]
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 191, in <listcomp>
    transformed_segs = [transformation(seg) for seg in path]
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 308, in <lambda>
    transformation = lambda seg: transform(seg, tf)
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 339, in transform
    return Arc(new_start, radius=new_radius, rotation=curve.rotation + rot,
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 1448, in __init__
    self._parameterize()
  File "/usr/lib/python3/dist-packages/svgpathtools/path.py", line 1503, in _parameterize
    raise ValueError("No such elliptic arc exists.")
```

BTW: the svgpathtool dependency pulls at least 64 MB of stuff, I think this is an overkill to implement `--shrink`

yaqwsx, 2022-01-18:
This is bug, I will look into it soon.

About svgpathtool dependency. It, unfortunatelly, pulls scipy in setup.py, which is huge, although, the dependency is optional. So if you manage to install without scipy, it still should work and be small. If you have a tip for substitute library, I will welcome such a tip. However, I cannot afford to implement another solution, e.g., via interpreting SVG curves of filtering the entities in KiCAD based on their visibility in the rendering.

yaqwsx, 2022-01-18:
This is actually a duplicate of #88. Your board has pads outside the board edge:

[image]

Due to #88, they get rendered and the image is therefore enlarged due to their presence. When you remove these pads, shrinking works.

closed 2022-01-18 by yaqwsx

yaqwsx, 2022-01-30:
A workaround for #88 was implemented as the bug probably won't be fixed in KiCAD.