yaqwsx/PcbDraw #85: PCBDraw does not account for the soldering paste

joric, 2022-01-03:
I want to know where I have soldering paste on the PCB. Kicad 3d view shows that, KicadScripts support that, how to do the same in PCBDraw? KicadScripts example output (note SMD pads are silver but the through-holes are golden):

[image]

yaqwsx, 2022-01-03:
At the moment, PcbDraw doesn't support the paste layer. The original motivation for PcbDraw is to create a nice drawing of the board, e.g., for pinout diagrams. Therefore, having a paste layer seemed unnecessary. 

If there are valid use-cases for having SMD pads rendered differently, we can add such a feature. But I don't want to implement a feature with limited use-case.

Nevertheless, if you submit a PR adding this feature, I will be happy to accept it. You basically have to modify the function `get_board_substrate` an extend the plot plan.

joric, 2022-01-03:
The valid case is a board preview in presentation, seeng soldering paste helps to spot issues and fix the board. I'll see what I can do. It perhaps would require fixing json templates and adding another entry (or maybe not ultimately, since all soldering paste is grey).

yaqwsx, 2022-03-01:
An update: I am considering a major rewrite of PcbDraw that should allow for custom plot plans. This would nicely address this feature request.

yaqwsx, 2022-03-16:
Implemented in fead173.

closed 2022-03-16 by yaqwsx