yaqwsx/PcbDraw #88: KiCAD 6.0 vias rendered as silkscreen

kb-elmo, 2022-01-14:
It seems that PCBdraw is rendering vias from KiCAD 6 files as silkscreen now.

Tested it by setting the silkscreen color to red and got this as a result.
[image]

Is this intentional?
Imho it does look kind of weird when all the vias are white.

Ideally should have the color of the board or be transparent like other holes.

yaqwsx, 2022-01-15:
Could you provide me with the source board so I can test it and debug it?

kb-elmo, 2022-01-15:
The problem happened on every KiCAD 6 PCB that I tested so far.

But here's the file that I used for the render above: [pcb.zip](https://github.com/yaqwsx/PcbDraw/files/7874704/pcb.zip)

yaqwsx, 2022-01-18:
I traced it that there is indeed a change in KiCAD 6 where the vias are included in generated SVG silkscreen files. However, I wasn't able to locate which settings cause that.

Electro707, 2022-01-23:
What's strange is that using KiCAD's built in SVG exporter, which uses the same class as pcbdraw fundamentally from what I could tell from KiCAD's code, exporting only the silkscreen layer doesn't show any vias:
[image]

Electro707, 2022-01-23:
Another thing I noticed is the exported from KiCAD's plotter directly colors vias white, while actual silkscreen black
[image]
[image]

yaqwsx, 2022-01-23:
Oh, sorry, I didn't post it here. This is definitely a bug in KiCAD, I opened an issue for that a few days ago: https://gitlab.com/kicad/code/kicad/-/issues/10491

We could make a work-around, but I would (slightly) prefer a fix in KiCAD.

yaqwsx, 2022-01-30:
Since the bug got low-priority I guess it won't be fixed until v7 (with the new API). Therefore, I implemented a workaraund in f1b0873.

closed 2022-01-30 by yaqwsx