yaqwsx/PcbDraw #89: Warning: component ... was not found

bzgec, 2022-01-18:
I am having a problem that no component is found.

```
$ pcbdraw voltLink.kicad_pcb test.png
------------------------------------------------------------------
Warning: component 'Fiducial_0.5mm_Mask1mm' for footprint 'FID1' from library 'Fiducial' was not found
Warning: component 'Fiducial_0.5mm_Mask1mm' for footprint 'FID2' from library 'Fiducial' was not found
Warning: component 'Fiducial_0.5mm_Mask1mm' for footprint 'FID3' from library 'Fiducial' was not found
Warning: component 'MountingHole_2.5mm_Pad' for footprint 'H1' from library 'MountingHole' was not found
Warning: component 'MountingHole_2.5mm_Pad' for footprint 'H2' from library 'MountingHole' was not found
Warning: component 'MountingHole_2.5mm_Pad' for footprint 'H3' from library 'MountingHole' was not found
Warning: component 'MountingHole_2.5mm_Pad' for footprint 'H4' from library 'MountingHole' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R10' from library 'Resistor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C3' from library 'Capacitor_SMD' was not found
Warning: component 'D_0603_1608Metric_Pad1.05x0.95mm_HandSolder' for footprint 'D2' from library 'Diode_SMD' was not found
Warning: component 'D_SMA' for footprint 'D1' from library 'Diode_SMD' was not found
Warning: component 'SOT-363_SC-70-6' for footprint 'Q1' from library 'Package_TO_SOT_SMD' was not found
Warning: component 'QFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm' for footprint 'U2' from library 'Package_DFN_QFN' was not found
Warning: component 'SOT-363_SC-70-6' for footprint 'U1' from library 'Package_TO_SOT_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C2' from library 'Capacitor_SMD' was not found
Warning: component 'kibuzzard-61E57FAE' for footprint 'kibuzzard-61E57FAE' from library '' was not found
Warning: component 'USB_Micro-B_Molex-105017-0001' for footprint 'J3' from library 'myLib' was not found
Warning: component 'SolderJumper-2_P1.3mm_Bridged_RoundedPad1.0x1.5mm' for footprint 'JP2' from library 'Jumper' was not found
Warning: component 'PinHeader_1x08_P2.54mm_Vertical' for footprint 'J4' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R6' from library 'Resistor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C7' from library 'Capacitor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C9' from library 'Capacitor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C10' from library 'Capacitor_SMD' was not found
Warning: component 'Fuse_1206_3216Metric_Pad1.42x1.75mm_HandSolder' for footprint 'F1' from library 'Fuse' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C5' from library 'Capacitor_SMD' was not found
Warning: component 'PinHeader_1x03_P2.54mm_Vertical' for footprint 'JP1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C1' from library 'Capacitor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C6' from library 'Capacitor_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R8' from library 'Resistor_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R3' from library 'Resistor_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R1' from library 'Resistor_SMD' was not found
Warning: component 'PinHeader_1x06_P2.54mm_Horizontal' for footprint 'J1' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'SOT-23-5_HandSoldering' for footprint 'U3' from library 'Package_TO_SOT_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R9' from library 'Resistor_SMD' was not found
Warning: component 'JST_SH_BM06B-SRSS-TB_1x06-1MP_P1.00mm_Vertical' for footprint 'J2' from library 'Connector_JST' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R2' from library 'Resistor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C8' from library 'Capacitor_SMD' was not found
Warning: component 'D_0603_1608Metric_Pad1.05x0.95mm_HandSolder' for footprint 'D3' from library 'Diode_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R5' from library 'Resistor_SMD' was not found
Warning: component 'PinHeader_1x08_P2.54mm_Vertical' for footprint 'J5' from library 'Connector_PinHeader_2.54mm' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C4' from library 'Capacitor_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R4' from library 'Resistor_SMD' was not found
Warning: component 'R_0603_1608Metric' for footprint 'R7' from library 'Resistor_SMD' was not found
Warning: component 'C_0603_1608Metric' for footprint 'C11' from library 'Capacitor_SMD' was not found
```

Running Arch Linux with `pcbdraw-git` package.
I tried installing it with `pip`, but then some components were populated and others weren't.

KiCad version: 6.0.1
PcbDraw version: 0.9.0-1
Python version: 3.10.1

Electro707, 2022-01-23:
Try the follwing modified PKGBUILD
```
# Maintainer: thegala <thegala at disroot.org>

_pkgname=pcbdraw
pkgname=pcbdraw-git
pkgver=252.1c3e689
pkgrel=1
arch=('i686' 'x86_64')
pkgdesc="Convert your KiCAD boards into nice looking 2D drawings suitable for pinout diagrams."
url="https://github.com/INTI-CMNB/pcbdraw"
license=('Apache2.0')
depends=('python' 'kicad')
makedepends=('git' 'python' 'python-setuptools')
source=("${_pkgname}::git+https://github.com/INTI-CMNB/pcbdraw")
md5sums=('SKIP')
provides=('pcbdraw')
conflicts=('pcbdraw')

prepare() {
  cd "${srcdir}/${_pkgname}"
  git submodule update --init
}

pkgver()
{
  cd "$_pkgname"
  echo $(git rev-list --count HEAD).$(git rev-parse --short HEAD)
}

package() {
  cd $_pkgname
  python setup.py install --root="$pkgdir"
}
```

I simply added a prepare callback that updates the clone repo's submodules, which is required to get the footprints.

bzgec, 2022-01-24:
I see that you already commented on AUR `pcbdraw-git` package. 

This adds some footprints, for the missing ones I should report to [PcbDraw-Lib](https://github.com/yaqwsx/PcbDraw-Lib/tree/master/KiCAD-base)?

Electro707, 2022-01-24:
Yeah, why not. 

You can also attempt to draw them yourself if you are familiar with vector editing (I would recommend Inkscape for it)

closed 2022-01-24 by bzgec

Sampozzo, 2022-02-03:
Hello i have the same issue on linux mint. Can you help me to solve this? I'm new to phyton scripts and i installed pcbdraw and pinion by pip. Kicad is working good.

Thank you

Electro707, 2022-02-04:
@Sampozzo  I would create a new Issue with as much detail and logging as possible.

Sampozzo, 2022-02-06:
@Electro707 I just created a really simple pcb to test pcbdraw and i have the same error said in this post:

`pcbdraw test-prj/test-prj.kicad_pcb ./image.svg`

`Warning: component 'PinHeader_1x03_P2.00mm_Vertical' for footprint 'J2' from library 'Connector_PinHeader_2.00mm' was not found
Warning: component 'PinHeader_1x03_P2.00mm_Vertical' for footprint 'J1' from library 'Connector_PinHeader_2.00mm' was not found`

I'm on a fresh Ubuntu install with kicad 6

[image]

Electro707, 2022-02-07:
@Sampozzo That error exists because there is no graphic for those footprints in [pcbdraw-lib](https://github.com/yaqwsx/PcbDraw-Lib), they will need to be created for those footprints. If you replace the 2mm header with 2.54mm header, the issue should go away as they are in `pcbdraw-lib`

Sampozzo, 2022-02-07:
@Electro707 i have the same error with 2.54 footprints, i solved by passing the --libs param but i have a blank image with no error.

What now i uderstand is that pinion need bpcdraw that need pcbdraw-lib... and you need to create a lib for every custom footprint you have in your project. This is not clear for people like me that found a great project like pinion, and this is why i found this thread, need to learn a lot :)

Thank you for your help