yaqwsx/PcbDraw #94: Missing 3rdparty directory resulting in ERROR: Could not install packages in Windows 10

yconst, 2022-04-11:
Following the instructions to install in Windows 10 using `pip install PcbDraw` in the KiCad Command Prompt, I get the following:

    ERROR: Could not install packages due to an OSError: [WinError 2] The system cannot find the file specified: 
    'C:\\Users\\****\\Documents\\KiCad\\6.0\\3rdparty\\Python39'

It seems there is no `3rdparty` directory in KiCad

yaqwsx, 2022-04-11:
This seems like a problem in KiCAD installation - there is no Python interpreter provided by KiCAD.

yconst, 2022-04-11:
I went ahead and did a full upgrade to Kicad 6.0.4 -- all default options. Now the error is 

    ERROR: Could not install packages due to an OSError: [WinError 2] The system cannot find the file specified: 
    'C:\\Users\\****\\Documents\\KiCad\\6.0.4'

The 6.0.4 directory doesn't even exist... it seems something fishy is going on with the kicad directory naming. I'm not so advanced to dig through the KiCad configuration to solve this.

I understand this is not immediately related with PcbDraw, so feel free to close it.

yaqwsx, 2022-05-08:
Recently, there was an issue where the root cause was aggressive anti-virus software: https://gitlab.com/kicad/code/kicad/-/issues/11578

yaqwsx, 2022-06-26:
Was this resolved? Was it the aggressive anti-virus software?

closed 2022-09-12 by yaqwsx