yaqwsx/PcbDraw #97: Generated SVG image is not possible to open in browsers (image is empty, white page)

roman-dvorak, 2022-06-05:
Hi, I am trying to generate a PCB svg image with PcbDraw plot tool. Image that could be displayed in a web browser. 

The image that I generate using the command `pcbdraw plot file.kicad_pcb plot.svg` I am able to open in some image viewers including, for example, inkscape. When I open the file in a browser, I see a blank (all white) page. If I view the source code in the browser, I see the entire XML there.

If I create an SVG image in the inkscape and open it in the browser, the browser will open it without any problems.

There will probably be a problem with some svg tag. But I can't figure out which it could be. We tried it in ubuntu 21.04, with current firefox and chromium (the same result).

Is this condition known or is it a mistake?

Electro707, 2022-06-05:
It's probably as issue of the way KiCAD exports SVGs where the numbers are too high and browsers does not support them.

See https://gitlab.com/kicad/code/kicad/-/issues/10569, https://gitlab.com/kicad/code/kicad/-/issues/9976, and https://gitlab.com/kicad/code/kicad/-/issues/9447

yaqwsx, 2022-06-06:
Yes, this is just like Electro707 says. I attempted to convert the units, however, it requires fully parsing and interpreting the geometry in SVG file and it seemed like too much work with a lot of corner cases, so it is not included in the current version. However, we might give it another shot.

roman-dvorak, 2022-06-07:
Thank you for clarification!

set-soft, 2022-10-21:
Hi @yaqwsx !

> I attempted to convert the units

Not really necesary. You can just change the SVG precision used by KiCad. Just going from 6 (default) to 5 fixes the problem. I'm using 4 to allow using really big page sizes.

Here you can see how I patched PcbDraw code to use a configurable precision:
https://github.com/INTI-CMNB/KiBot/commit/417dbadcd2e0a7a6bf3c9a4b936532f85530ad77#diff-a3084f061bd46974faf8029e89b48d6c0ce04f85da549c1504d13710e13bfc00

closed 2022-10-24 by yaqwsx