yaqwsx/PcbDraw #99: [Suggestion] Eliminate engineering_number dependency

set-soft, 2022-06-24:
I see that 7b2a4b9bcc7c76ca689cb616563cccfed851612c introduced a new dependency `engineering_number`.

I think you should avoid it using something like: https://github.com/INTI-CMNB/PcbDraw/commit/74ccf4bd56f37c8f6a03f1b588014146afbde5a5

Which is a poor as engineering_number but solves the problem in 8 LOCs instead of pulling a new dependency.

As you are undergoing a major rewrite I'll suggest taking a look at https://github.com/INTI-CMNB/KiBot/blob/master/kibot/bom/units.py
This code comes from KiBoM and I think iBoM also uses it. It handle things like `1k5` or `1 meg`

yaqwsx, 2022-06-26:
I agree with this and I will address this.

yaqwsx, 2022-09-22:
Removed in 2635a07b12af3041f2b15d557bd3d780091bc7fa. It was replaced by a code from https://github.com/yaqwsx/jlcparts.

closed 2022-09-22 by yaqwsx