yaqwsx/PcbDraw #102: Graphical Errors on Edge.Cuts Rendering

duckyb, 2022-08-11:
This is a follow up of: #96 

Sometimes PcbDraw fails to render the Edge.Cuts layer correctly. This seems to be unrelated to the Edge.Cuts tolerance threshold, and more likely to happen with Kicad 6 files.

## First Example

PCB File: [battoota.zip](https://github.com/yaqwsx/PcbDraw/files/9307166/battoota.zip)

PcbDraw output: 

[image]

### Tolerances
 
This time the issue doesn't seem to come from the edge.cuts tolerance, as we can see from the raw file that all segments start/end match exactly:
```
  (gr_line (start 164.316185 -38.254903) (end 164.319791 -118.263254) (layer "Edge.Cuts") (width 0.05) (tstamp 181135d6-242b-4baf-94b0-054802ef6df0))
  (gr_arc (start 46.209791 -33.808254) (mid 43.066701 -35.110164) (end 41.764791 -38.253254) (layer "Edge.Cuts") (width 0.05) (tstamp 23a9b3df-ce2e-4f15-92a4-05c00d9cd2cc))
  (gr_arc (start 159.874791 -122.708254) (mid 163.017881 -121.406344) (end 164.319791 -118.263254) (layer "Edge.Cuts") (width 0.05) (tstamp 28d95701-1ce4-4407-9f61-1674332e1642))
  (gr_line (start 46.209791 -33.808254) (end 159.871185 -33.809903) (layer "Edge.Cuts") (width 0.05) (tstamp 5c946c69-aabf-45dc-9f47-f37983b2dc53))
  (gr_line (start 41.761149 -118.266118) (end 41.764791 -38.253254) (layer "Edge.Cuts") (width 0.05) (tstamp 7b32ef33-8c7b-417f-9260-1a8773398f8f))
  (gr_line (start 159.874791 -122.708254) (end 46.206149 -122.711118) (layer "Edge.Cuts") (width 0.05) (tstamp d92cfbfa-da4b-4f63-8ad6-7bb6977d4f44))
  (gr_arc (start 41.761149 -118.266118) (mid 43.063059 -121.409208) (end 46.206149 -122.711118) (layer "Edge.Cuts") (width 0.05) (tstamp def56ef8-2877-4427-9903-57250c5a3b07))
  (gr_arc (start 164.316185 -38.254903) (mid 163.014275 -35.111813) (end 159.871185 -33.809903) (layer "Edge.Cuts") (width 0.05) (tstamp f52d2bef-e44b-4029-9335-e50fd8ba2057))
```
## Second Example

I have also noticed that on some PCB files, the issue appears after upgrading the file to Kicad 6. Here is an example of that.

PCB File (kicad 5): [paroxysm_left.zip](https://github.com/yaqwsx/PcbDraw/files/9307209/paroxysm_left.zip)

### BEFORE upgrading to Kicad 6

[image]

### AFTER upgrading to Kicad 6

[image]

cc: @AlaaSaadAbdo

set-soft, 2022-08-11:
Hi @duckyb !
Just my 2 cents, I think this *is* a rounding issue, I tried using my fork of PcbDraw and got correct bounds. Note that my fork is just an stable PcbDraw with the fix for the rounding issue. You won't see the rounding issue in the PCB file, the main problem is in the SVG exporter, PcbDraw uses the output of the SVG Plotter. With the patch applied you'll get correct borders.

[image]
[image]
[image]
[image]

AlaaSaadAbdo, 2022-08-11:
Hi set-soft,

Is your fork available?
I couldn't find it in the list of forks.

If you could share it, that would be much appreciated.

Cheers!
Alaa

duckyb, 2022-08-11:
@set-soft Thank you for looking into this for us. We've tried your fork too and it worked. It's a shame that this fix isn't part of kikit's dockers but we'll use the fork for now.

closed 2022-08-11 by duckyb

set-soft, 2022-08-11:
BTW @duckyb and @AlaaSaadAbdo the patched PcbDraw is used for [KiBot](https://github.com/INTI-CMNB/KiBot/) and is available in its GitHub Action and docker images are available. 
Lamentably I didn't have the time to include KiKit, but is in my TODO list.

yaqwsx, 2022-08-18:
Hi, KiKit's docker includes always the latest released version of PcbDraw.

PcbDraw code has changed a lot before this bug was reported, unfortunately. For the sake of simplicity of maintenance, I only move forward, and I don't track divergent branches for patching existing releases. Since the code brings a lot of breaking changes, it has to be a major release. Nevertheless, I didn't manage to release it on schedule (end of June).

The v1 release of PcbDraw is near (I hope in a couple of months). What's left is a tool to manage its library. Version 1 fixes some of the technical debt we had and also allows for custom plot layouts, and it can even export rendered pictures of the board.

yaqwsx, 2022-09-26:
FYI: v1 was released.