yaqwsx/PcbDraw #103: Support mistune >= 2.0.0

dvzrv, 2022-08-29:
Hi! This project is packaged on Arch Linux and we would like to be able to get rid of our python-mistune1 package (which ships mistune < 2.0.0).

It would be awesome if pcbdraw could work with current mistune (https://pypi.org/project/mistune/) which is >= 2.0.0 as that would allow us to move on :)

yaqwsx, 2022-08-29:
Hi, I can arrange that with the upcoming v1 release.

pintert3, 2022-09-01:
I agree with this, if I'm also willing to help if I get time

dvzrv, 2022-09-03:
@yaqwsx Thanks. As I have no idea how your codebase works and the renderer code seems pretty custom, I don't think I have the time to spend on this unfortunately. It doesn't look overly complex though, so if @pintert3 or anyone else is up for the task, that'd be awesome! :)

yaqwsx, 2022-09-05:
Just a note: the only place where mistune is used is in the populate functionality located in https://github.com/yaqwsx/PcbDraw/blob/master/pcbdraw/populate.py

What we do is we add a new tag into the markdown parser - the one that renders a board with new components added.

pintert3, 2022-09-06:
Thanks, that'll help a lot

yaqwsx, 2022-09-11:
I merged the PR. I will take a week or two for final testing, and then I will release version 1.0.0 of PcbDraw.

yaqwsx, 2022-09-22:
v1 was released, closing.

closed 2022-09-22 by yaqwsx