yaqwsx/PcbDraw #104: Kicad 6.99 - Unknown layer ID

cchaine, 2022-08-30:
Hi,

I have encountered this issue while trying to use PcbDraw with nightly version `6.99.0-3148-gdb6c153ad9-dirty` of Kicad.
```
wx._core.wxAssertionError: C++ assertion "false" failed at /home/ubuntu/tools/kicad/common/layer_id.cpp(190) in LayerName(): Unknown layer ID -1

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/pcbdraw", line 11, in <module>
    load_entry_point('PcbDraw==0.9.0+36.g7118c48.dirty', 'console_scripts', 'pcbdraw')()
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/ui.py", line 174, in plot
    image = plotter.plot()
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 1028, in plot
    plotter.render(self)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 650, in render
    self._plotter.execute_plot_plan(to_plot)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 1182, in execute_plot_plan
    pctl.OpenPlotfile(action.name, pcbnew.PLOT_FORMAT_SVG, action.name)
  File "/usr/local/lib/python3/dist-packages/pcbnew.py", line 7173, in OpenPlotfile
    return _pcbnew.PLOT_CONTROLLER_OpenPlotfile(self, *args)
SystemError: <built-in function PLOT_CONTROLLER_OpenPlotfile> returned a result with an error set
```

This happens when calling `OpenPlotFile` where kicad calls [`StartPlotBoard`](https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/pcbplot.cpp#L474) which then calls [`ToLayerId`](https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/plot_board_layers.cpp#L1145) passing it the layer attribute of the plot controller. At the time of calling the function, the layer attribute hasn't been initialised yet and is equal to `-1`.

This issue should be resolved in Kicad in my opinion (see [here](https://gitlab.com/kicad/code/kicad/-/issues/12314)), but a workaround could be to initialize the layer attribute manually before calling OpenPlotFile.

yaqwsx, 2022-08-30:
Hi, similarly to my other projects, I won't support 6.99 until late fall/early winter 2022. The KiCAD API is highly unstable between major releases, and trying to support the nightly until the API is stable is just wasted work. 

Thank you for reporting the issue to KiCAD!

cchaine, 2022-08-30:
No worries that makes sense. I have managed to fix the issues related to the nightly and I'll post a MR for others who would wan't to use it.

cchaine, 2022-08-30:
PR #106 should resolve this issue.

cchaine, 2022-09-12:
Fixed upstream with commit [95874aee](https://gitlab.com/kicad/code/kicad/-/commit/95874aeeb48a5b58d106d5e3237cbfd6709567bc).

closed 2022-09-12 by cchaine

guigur, 2022-10-04:
just updated kicad to version 6.0.8 and had the same issue

```
Application: KiCad

Version: 6.0.8-f2edbf62ab~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.15.0-48-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Sep 30 2022 16:52:13
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.85.0
	ngspice: 36
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

error logs:
```
wx._core.wxAssertionError: C++ assertion "false" failed at /build/kicad-YdG4kW/kicad-6.0.8-0/kicad/common/layer_id.cpp(180) in LayerName(): Unknown layer ID -1

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/garthaud/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(run())
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/ui.py", line 174, in plot
    image = plotter.plot()
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1053, in plot
    plotter.render(self)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 674, in render
    self._plotter.execute_plot_plan(to_plot)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1207, in execute_plot_plan
    pctl.OpenPlotfile(action.name, pcbnew.PLOT_FORMAT_SVG, action.name)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 7586, in OpenPlotfile
    return _pcbnew.PLOT_CONTROLLER_OpenPlotfile(self, aSuffix, aFormat, aSheetDesc)
SystemError: <built-in function PLOT_CONTROLLER_OpenPlotfile> returned a result with an error set
```

I've tried to uninstall the stable version with pip and install the github version with no changes.

guigur, 2022-10-04:
Just updated Kicad to version 6.0.8 and had the same issue

Kicad build:
```
Application: KiCad

Version: 6.0.8-f2edbf62ab~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.15.0-48-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Sep 30 2022 16:52:13
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.85.0
	ngspice: 36
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

PcbDraw error logs:
```
wx._core.wxAssertionError: C++ assertion "false" failed at /build/kicad-YdG4kW/kicad-6.0.8-0/kicad/common/layer_id.cpp(180) in LayerName(): Unknown layer ID -1

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/garthaud/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(run())
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/ui.py", line 174, in plot
    image = plotter.plot()
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1053, in plot
    plotter.render(self)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 674, in render
    self._plotter.execute_plot_plan(to_plot)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1207, in execute_plot_plan
    pctl.OpenPlotfile(action.name, pcbnew.PLOT_FORMAT_SVG, action.name)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 7586, in OpenPlotfile
    return _pcbnew.PLOT_CONTROLLER_OpenPlotfile(self, aSuffix, aFormat, aSheetDesc)
SystemError: <built-in function PLOT_CONTROLLER_OpenPlotfile> returned a result with an error set
```

cchaine, 2022-10-04:
Hi, have you tried using PR #106 of pcbdraw ?

guigur, 2022-10-04:
I've just tried your PR and summoned a different error

```
Traceback (most recent call last):
  File "/home/garthaud/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(run())
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/garthaud/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/ui.py", line 174, in plot
    image = plotter.plot()
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1027, in plot
    plotter.render(self)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 649, in render
    self._plotter.execute_plot_plan(to_plot)
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 1192, in execute_plot_plan
    action.action(action.name, os.path.join(tmp, svg_file))
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 687, in _process_outline
    for hole in collect_holes(self._plotter.board):
  File "/home/garthaud/.local/lib/python3.8/site-packages/pcbdraw/plot.py", line 596, in collect_holes
    orientation=pad.GetOrientation().AsDegrees(),
AttributeError: 'float' object has no attribute 'AsDegrees'
```

cchaine, 2022-10-04:
I'm not too surprised as EDA_ANGLES have not yet been introduced into the stable branch.

As a workaround, you could add [this line](https://github.com/yaqwsx/PcbDraw/pull/106/files#diff-bab71f1426a1fbcd6c07c95c7499a7cf0f8b48d777ee74b99f5ad4dd8e4d17bcR1175) to the version of PcbDraw you used in the first place.

guigur, 2022-10-04:
You are the best!
I've changed the line you told me about and forked the repo for anyone interested in your solution.
https://github.com/guigur/PcbDraw-kicad-6.0.8

cchaine, 2022-10-04:
No worries 😁

set-soft, 2022-10-05:
Just my 2 cents: the way PcbDraw uses the plot controller is wrong, you should open it once for each generated file, and also close it. In this way you set the layer **before** the open. I took a look to the two places where KiBot uses the plot controller and I'm setting the layer before the open. I'm also doing a close for each open.

cchaine, 2022-10-05:
I agree with you, however you might want to plot multiple layers within one single file. Then it might not make much sense to set the first layer to be drawn before opening the file.

Kicad appears to agree with this as [this](https://gitlab.com/kicad/code/kicad/-/commit/95874aeeb48a5b58d106d5e3237cbfd6709567bc) issues has been merged upstream.

yaqwsx, 2022-10-05:
> Just my 2 cents: the way PcbDraw uses the plot controller is wrong, you should open it once for each generated file, and also close it. In this way you set the layer **before** the open. I took a [...]

Could you direct me to the source that indicates that the plot controller cannot be reused? [The official example](https://gitlab.com/kicad/code/kicad/-/blob/master/demos/python_scripts_examples/plot_board.py) reuses the plot controller and also claims that a setting layer beforehand should be only mandatory for gerbers. But I guess there are changes in 6.0.8 that do not agree with the claims in the example. Therefore, I added e009f57 to address this.

set-soft, 2022-10-05:
> Could you direct me to the source that indicates that the plot controller cannot be reused? [The official [...]

Is just my experience using KiCad 5 and 6.