yaqwsx/PcbDraw #105: Kicad 6.99 - EDA_ANGLE compatibility

cchaine, 2022-08-30:
Hi,

I have encountered this issue while trying to use PcbDraw with nightly version `6.99.0-3148-gdb6c153ad9-dirty` of Kicad.

```
Traceback (most recent call last):
  File "/usr/local/bin/pcbdraw", line 11, in <module>
    load_entry_point('PcbDraw==0.9.0+36.g7118c48.dirty', 'console_scripts', 'pcbdraw')()
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click-8.1.3-py3.8.egg/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/ui.py", line 174, in plot
    image = plotter.plot()
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 1028, in plot
    plotter.render(self)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 650, in render
    self._plotter.execute_plot_plan(to_plot)
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 1194, in execute_plot_plan
    action.action(action.name, os.path.join(tmp, svg_file))
  File "/usr/local/lib/python3.8/dist-packages/PcbDraw-0.9.0+36.g7118c48.dirty-py3.8.egg/pcbdraw/plot.py", line 696, in _process_outline
    position[0], position[1], -hole.orientation / 10)
TypeError: bad operand type for unary -: 'EDA_ANGLE'
```

This is because Kicad introduced a type called EDA_ANGLE. This can be resolved by using the `AsDegrees()` member function of the EDA_ANGLE class.

cchaine, 2022-08-30:
PR #106 should resolve this issue.

yaqwsx, 2023-11-05:
We already support KiCAD 7 with PcbDraw 1, closing.

closed 2023-11-05 by yaqwsx