yaqwsx/PcbDraw #111: populate: highlighting rectangle is overlapping

dhaillant, 2022-09-05:
Hello,
In populate, the highlighting rectangle is not centered around the component and is bigger than the component footprint, and then when two components are adjacent, the rectangles are overlapping.

[image]

pcbdraw, version 0.9.0+36.g7118c48

yaqwsx, 2022-09-12:
Could you please test if the most recent version (with changes from today) fixes the problem?

dhaillant, 2022-09-12:
Thank you for your help!
Indeed the offset is now gone. The red zone is now correctly centered around the component:
[image]

However, the red zone is still a bit "too large" (I know this is a matter of taste probably), but as you can see on the picture above, when two adjacent components are to be populated in the same step, the red zone is overlapping and the color (as it's semi-transparent) is a bit darker on the overlap...

Then, on following steps, as the red zone overlaps adjacent components, it creates some visual confusion I think:
[image]

The components on this PCB are placed on a 2.54mm grid, so, pretty common spacing I guess.

So I'd like to know if there's a way to reduce the red zone size, or make it just the size of the footprint, please?

Thank you very much.

yaqwsx, 2022-09-12:
The size is specified by the `highlight-padding` property of board style.

dhaillant, 2022-09-12:
Thank you, that did the trick!
I first thought that the json style files were for the PCB rendering only and that populate wasn't concerned.
Thanks again!

closed 2022-09-12 by dhaillant