yaqwsx/PcbDraw #119: Python can't find KiCad python libs.

CapnKernel, 2022-09-29:
I'm running KiCad 6.0.7 in Fedora Linux 36.  I have created a virtual env with `python -m venv venv`, activated the venv, and run `pip install PcbDraw`.  When I run `pcbdraw --help`, I get the message `ModuleNotFoundError: No module named 'pcbnew'`.  I'm guessing this is because my invocation of python can't find KiCad's Python libraries.

If I start Python in this venv, this is the value of `sys.path`:
```
['', '/usr/lib64/python310.zip', '/usr/lib64/python3.10', '/usr/lib64/python3.10/lib-dynload', '/home/mjd/git/ozhelm/venv/lib64/python3.10/site-packages', '/home/mjd/git/ozhelm/venv/lib/python3.10/site-packages']
```
If I start Python via KiCad (ie, the KiPython shell), I get this value for `sys.path`:
```
['', '/usr/lib/python3.10/site-packages', '/usr/lib64/python310.zip', '/usr/lib64/python3.10', '/usr/lib64/python3.10/lib-dynload', '/usr/lib64/python3.10/site-packages', '/usr/share/kicad/scripting', '/usr/share/kicad/scripting/plugins', '/home/mjd/.local/share/kicad/6.0/scripting', '/home/mjd/.local/share/kicad/6.0/scripting/plugins', '/usr/share/kicad/scripting']
```
I don't know if this is what should be picked up, but there's a `pcbnew` in `/usr/lib/python3.10/site-packages/`.

Here's the list of files in this version of Fedora's KiCad: 

  https://rpmfind.net/linux/RPM/fedora/updates/36/x86_64/Packages/k/kicad-6.0.7-1.fc36.x86_64.html

KiPython correctly picks up `/usr/lib/python3.10/site-packages`.

How can I run `pcbdraw` so it finds `pcbnew`?

CapnKernel, 2022-10-05:
Can I get some help on this please?  Thank you.

CapnKernel, 2022-10-08:
Would really love some help

yaqwsx, 2022-10-09:
Can you import pcbnewTransition from the python REPL inside pcbnew?

CapnKernel, 2022-10-09:
I tried starting a shell in pcbnew using Tools > Scripting Console.
```
Python 3.10.7 (main, Sep  7 2022, 00:00:00) [GCC 12.2.1 20220819 (Red Hat 12.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/mjd/.config/kicad/6.0/PyShell_pcbnew_startup.py
>>> 
>>> import pcbnewTransition
Traceback (most recent call last):
  File "/usr/lib64/python3.10/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'pcbnewTransition'
>>> 
>>> import sys
>>> sys.path
['', '/usr/lib/python3.10/site-packages', '/usr/lib64/python310.zip', '/usr/lib64/python3.10', '/usr/lib64/python3.10/lib-dynload', '/usr/lib64/python3.10/site-packages', '/usr/share/kicad/scripting', '/usr/share/kicad/scripting/plugins', '/home/mjd/.local/share/kicad/6.0/scripting', '/home/mjd/.local/share/kicad/6.0/scripting/plugins', '/usr/share/kicad/scripting']
>>> 
```
pcbnew was started from the KiCad main shell, which was started from my desktop's start menu.
pcbdraw is installed in a venv inside the directory of the project KiCad currently has open.  There's a `venv/lib/python3.10/site-packages/pcbnewTransition/__init__.py` file in that venv.

I don't know how to get the python interpreter I start in the venv outside of KiCad to know about KiCad's libraries, and I don't know how to get the python interpreter I start from pcbnew to know about the pcbdraw libraries.

CapnKernel, 2022-10-11:
Ahh I see.  Your one line message gave me the clue I needed to find other places on the internet that have talked about this problem.

The solution is to create the venv with the `--system-site-packages` flag, ie:
```
python -m venv --system-site-packages venv
```
Then python started in the venv will look at the packages in the system location, which is where KiCad puts its python code.

Hope this helps someone else!

yaqwsx, 2022-10-24:
Thanks for sharing the solution.

closed 2022-10-24 by yaqwsx