yaqwsx/PcbDraw #129: Boards with complex outline renders broken

korken89, 2022-10-11:
Hi, thanks for an awesome tool!

I've been using this tool to render PCBs for a while, and now I was going to start rendering my panels generated with KiKit.
However the panels comes out looking like below, but it works fine in KiCAD and it's 3D viewer (that is generally very sensitive to board edge errors).
So I'm quite sure the board edge is fine.

[image]

yaqwsx, 2022-10-11:
You can share it privately via email: email@honzamrazek.cz. I will take a look at it and investigate what's wrong.

yaqwsx, 2022-10-11:
The problem is not present in v1.0; the image was sent via e-mail.

closed 2022-10-11 by yaqwsx