yaqwsx/PcbDraw #132: ModuleNotFoundError: No module named 'pyvirtualdisplay'

Khalinda, 2022-11-05:
I was running KiCAD 6.0.7 and had previously installed KiKit and PcbDraw.  I just upgraded to KiCAD 6.0.9 and I received a warning that KiKit had to be reinstalled.  That was successful, but when I tried to install PcbDraw, following the instructions in the Installation Guide, I obtained the error message in the issue title.  I am running on Windows 11, I have Inkscape 1.2.1 installed.

Khalinda, 2022-11-05:
I just tried "pip install pyvirtualdisplay".  That seems to resolve the problem.

orobardet, 2022-11-06:
Had the same issue: Kicad 6.0.9, on Windows, no PcbDraw plugin installed. I installed it on kicad's command prompt with `pip install PcbDraw`, and I got the same error about missing pyvirtualdisplay when trying to run PcbDraw.

I can confirm installing the missing package manually with `pip install pyvirtualdisplay` solves the issue.

Maybe pyvirtualdisplay needs to be added as a dependency in setup.py?

yaqwsx, 2022-11-07:
`pyvirtualdisplay` is intentionally not installed on Windows as the package doesn't work on Windows. However, there was a problem that we imported in (when we shouldn't). This is fixed in [942566e](https://github.com/yaqwsx/PcbDraw/commit/942566e957f1011b402af3dfb84d0c76cec1ad16).

closed 2022-11-07 by yaqwsx