yaqwsx/PcbDraw #133: Unknown layer ID -1

Khalinda, 2022-11-05:
I just ran PcbDraw for the first time, so maybe I'm doing it wrong.  I'm running Windows 11 with KiCAD 6.0.9.  I opened a KiCAD command window and I entered a command like:

pcbdraw plot "test.kicad_pcb" "test.svg"

I obtained the following error message:

wx._core.wxAssertionError: C++ assertion ""false"" failed at ..\..\common\layer_id.cpp(180) in LayerName(): Unknown layer ID -1

PcbDraw got as far as creating a minimal .SVG file n a temp directory.

yaqwsx, 2022-11-07:
Could you share the source board? All tests are passing and I struggle to reproduce this.

Khalinda, 2022-11-07:
Here is a board that was created in KiCAD 5.x.  I pulled it into KiCAD 6.0.9 to convert it to 6.x then I ran PcbDraw and I obtained the ID -1 error.  Do you need anything other than the kicad_pcb file?
[Card.zip](https://github.com/yaqwsx/PcbDraw/files/9953795/Card.zip)

closed 2022-11-08 by yaqwsx

Khalinda, 2022-11-08:
Thanks.  That works.