yaqwsx/PcbDraw #135: No footprints

Andy79881, 2022-11-13:
I've got the following messages:
Component Connector_PinHeader_2.54mm:PinHeader_1x05_P2.54mm_Horizontal has not footprint.
Component LED_SMD:LED_0805_2012Metric_Pad1.15x1.40mm_HandSolder has not footprint.
Component Connector_PinHeader_1.27mm:PinHeader_2x07_P1.27mm_Vertical has not footprint.
Component Connector_PinHeader_2.54mm:PinHeader_2x02_P2.54mm_Vertical has not footprint.
Component TestPoint:TestPoint_Keystone_5000-5004_Miniature has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder has not footprint.
Component Crystal:Crystal_SMD_HC49-SD has not footprint.
Component Package_QFP:LQFP-48_7x7mm_P0.5mm has not footprint.
Component Button_Switch_SMD:SW_SPST_B3S-1000 has not footprint.
Component Capacitor_SMD:CP_Elec_4x5.8 has not footprint.
Component TestPoint:TestPoint_Keystone_5000-5004_Miniature has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_1210_3225Metric_Pad1.33x2.70mm_HandSolder.back has not footprint.
Component Package_TO_SOT_SMD:SOT-23.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Crystal:Crystal_SMD_3215-2Pin_3.2x1.5mm.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Connector_PinHeader_2.54mm:PinHeader_1x16_P2.54mm_Vertical.back has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Resistor_SMD:R_0805_2012Metric_Pad1.20x1.40mm_HandSolder.back has not footprint.
Component Capacitor_SMD:C_0805_2012Metric_Pad1.18x1.45mm_HandSolder.back has not footprint.
Component Connector_PinHeader_2.54mm:PinHeader_1x16_P2.54mm_Vertical.back has not footprint.
Component Package_TO_SOT_SMD:SOT-223-3_TabPin2.back has not footprint.
Component Inductor_SMD:L_0805_2012Metric_Pad1.05x1.20mm_HandSolder.back has not footprint.
Component Diode_SMD:D_SOD-123.back has not footprint.

According to the usage page: "...plotting them. The result is a simplified and stylized board image. However, you have to supply hand-drawn footprint images. It can also remap symbols, easily select board style and also, render resistor values color bands."
If I'm looking to the examples there are footprints shown. How can I get the footprints plotted?

Andy79881, 2022-11-14:
I've got it somehow partly working with the remap.json file. But I have to enter each component there and it is error prone. Furthermore not all fottprints will be displayed even if they are already used for another reference designator.

e.g. extract fom remap.json:
	"R6": "Resistor_SMD:R_0805_HandSoldering",
    "R7": "Resistor_SMD:R_0805_HandSoldering",
	"R2": "Resistor_SMD:R_0805_HandSoldering",
	"R3": "Resistor_SMD:R_0805_HandSoldering",

SVGs for RefDes R7 and R6 are displayed, for R2 and R3 not. Another issue is the wrong orientation of symbols as pin headers or SOT-23 packages. For me the whole procedure takes to much time and it is really annoying.

Electro707, 2022-11-20:
Could you try updating `pcbdraw` then running your script (ideally from git directly instead of a release). When I downloaded your SVG it looks black, even in Inkscape
[image]

Unfortunately what happened with the resistor mapping is that in previous versions of KiCAD the resistor footprint used to be called `R_0805_HandSoldering` but that has changed in V6. There have been discussions as to handling cases like this without going thru and renaming files, such as in #66 and https://github.com/yaqwsx/PcbDraw-Lib/issues/24

Could you send the board files if possible? It looks like it _might_ be a bug with how components are handled when placed on the backside of the board.

Andy79881, 2022-11-21:
Board file can be found in my github repository. 
https://github.com/Andy79881/STM32L051-dev-board

Andy79881, 2022-11-21:
I have tried to install the pcb draw under windows as follows:
pip3 install git+https://github.com/yaqwsx/PcbDraw@master
This works basically but I got error messages if the remap.json file is handled. After re-installing release v1.01 it works for me as before. I do not know what the issue with the git version is.

Electro707, 2022-11-21:
Hello,
I have taken a look at your board file, and have come to the following conclusions so far:
- The reason the SOT32 is oriented wrong is because you are using the Eagle version of the footprint, in which it's orientation is different from KiCAD's. We would need to create a SOT23 drawing (or at least migrate the one from Eagle) to KiCAD. I might do that on my next drawing PR as I plan on creating some SOT related drawing such as the SOT223
- Can confirm that the pin header is flipped by 180. This also caused issue other test components when placed on the backside, such as a certain staggered LED and QFP package.

Regarding PcbDraw not working with the git version, that is bizzare to me as I am currently using the master branch codebase. Can you send the error code when you tried using the master branch?

Bug with QFP package from flipping it
[image]
Improper Staggered LED from the backside:
[image]
Proper from the front:
[image]