yaqwsx/PcbDraw #137: [Suggestion] Just a thought

mundodisco8, 2023-01-11:
Hi,

I discovered PcbDraw through KiBot. I played a bit with it and thought that it was a pretty cool idea, especially because I was actually thinking in making an assembly manual for a board I'm making and this tool would help me massively with it.

As I said, I played a bit with the `populate` side of the tool and then realised the amount of work that would be required to get my components into the shared library, and not only that, but the coordination required to get something that is usable when multiple people work on it: things like getting outline sizes consistent across different parts, or everyone using the same palette of colours (so we all use the same grey for "steel" parts, for example, or the same orange shade for copper, brass or gold).

And then I thought, wouldn't it be just easier to play with the 3D viewer and take different snapshots with different components on it? My idea, and I don't know how feasible this would be, is to make a copy of the board file, set all component types to "Through Hole" for example, and enable only showing "SMD" components. Take a snapshot from the 3D viewer, and then change "through hole" components to "SMD" as required.

I'm pretty sure someone has thought about this before, and it might be against the general goal of the project, so forgive me if that's the case.

yaqwsx, 2023-01-11:
This was heavily discussed in https://github.com/yaqwsx/PcbDraw/issues/9. TLDR: It is wanted to have drawn images and renders, but it is technically impossible at the moment due to the limitations of KiCAD.

With the current KiCAD API, it is impossible to properly capture rendered board. The `render` command of PcbDraw is rather a big hack.

To some extent, the idea of 3D renders was deployed in https://github.com/yaqwsx/Pinion - but it works there only because we need only a single image, not tenths of them which would cause the manual to render for unacceptable times.

Until we have a proper API in KiCAD, I refuse to invest development time into half-functional hacks and only the plotted style will be supported for populate.

set-soft, 2023-01-11:
For @mundodisco8 and @yaqwsx:

I created a post in the [KiCad foum](https://forum.kicad.info/t/populate-pcdraw-now-using-3d-rendered-boards-kibot-kiauto/38350) showing a comparisson between both methods.
For brevity [here](https://set-soft.github.io/kibot_populate_example/Generated/Render_3D/index.html) is the `populate` output using `render_3d`. More information in the post and the repo.

yaqwsx, 2023-01-11:
@set-soft Thank you for letting me know! How long does it take to render the populate manual? Does it work on Windows?

yaqwsx, 2023-01-12:
The times seem usable. Good job!

Regards Windows: Most potential users for Populate refuse/are unable to setup CI, so they require it to run locally. Often, setting up WSL and having two instances of KiCAD is unacceptable for them.

set-soft, 2023-01-12:
> Often, setting up WSL and having two instances of KiCAD is unacceptable for them.

Most Windows users of KiBot does it. But I think docker is much more easy, not sure about performance issues.
With docker images is just some disk space. KiCad wastes too much disk, the 3D models are 5 GiB, much less than a docker image of KiBot, including KiCad. This is because KiBot downloads official 3D models on-the-fly (from the KiCad repo). So you can get a very functional KiBot image using just 1.3 GiB. If you need all the tools it will jump to 2.8 GiB.

WSL: I think it will become more and more common for people to enable it. Microsoft was forced to add some Linux support because of users preassure. First was an add-on, then something you just need to enable ...
Now is so easy that more and more packages will ask you to just enable WSL.
Installing KiBot on Debian/Ubuntu is trivial, and the Debian dependencies makes it very convenient. Installing the KiBot package will install KiCad and, if you ask for all suggested dependencies, will also install things like KiKit. No need to install things one by one.

Supporting KiAuto (the code that deals with contolling the KiCad GUI) on Windows in a reliable way means that somebody must implement the equivalent of the `interposer` lib. I don't know how hard/easy is this, I programmed for Windows loooong time ago and gave up, too broken. But I think this is something doable ([Microsoft Detour](https://github.com/microsoft/Detours)). Using the interposer the KiCad handling speed increased notably, and the reliability went from something like 95% to something around 99,9%. I even tried to render boards that takes 40 seconds to load on KiCad, with render times of minutes, and KiAuto was able to detect the KiCad state accurately, even when it tooks minutes to finish the render. The interposer lib looks like a hacking tool, it intercepts dynamic libs calls and patches their behavior. I was able to add accelerators to widgets that are hard to access in the GUI, or detect when KiCad opens and closes a file, all with 100% accuracy.

mundodisco8, 2023-01-19:
Hey @yaqwsx, thanks for your reply, sorry not the confusing subject. I understand the effort needed to do it. I'll close this ticket.

@set-soft, that's impressive, I'm going to give it a go. About the last image, is it the KiCad renderer or are you using something else?

closed 2023-01-19 by mundodisco8

set-soft, 2023-01-19:
> @set-soft, that's impressive, I'm going to give it a go. About the last image, is it the KiCad renderer or are you using something else?

This Blender, will be available soon on KiBot. In fact is available, but the docker images needs some adjusts.

set-soft, 2023-01-27:
Hi @yaqwsx and @mundodisco8:

Here is the populate example using [blender](https://set-soft.github.io/kibot_populate_example/Generated/Blender/index.html)

This is the top view for all components soldered:

[image]

Of course it has a price: almost 5 times slower.

The images were created using 10 samples, you can get more quality with more samples ... and more time

mundodisco8, 2023-01-27:
That's amazing! can't wait for it to be merged into master!

set-soft, 2023-01-28:
Hi @mundodisco8 !
You can get it using the `dev` docker images. These images are truly CI/CD. They contain all new stuff and fixes. When I work on a feature that will take various commits and break things I use another branch. The above picture was generated by a `dev` image. I'm also switching to use *full* images for GitHub Actions, so now things like PanDoc and Blender are available fro GHA. If the user wants to save download time must switch to regular images, not GHA.