yaqwsx/PcbDraw #140: Browser display issue, unexpected crop

adamws, 2023-02-26:
Hi, I'm facing issues when displaying svg's generated by PcbDraw in web browsers. Files are ok when opened with inkscape.
I'm using Kicad 6.09 and PcbDraw 1.0.1

Take for example this svg file:
[image]
there should be 9 elements in a row (github also renders 8 elements, at least for me):

This is how this looks in firefox v110:
[image]

and in chromium v109:
[image]

I checked both windows and linux. I didn't check other browsers.

After download (in firefox `Save Image As` option) and open in inkscape, it is as expected:
[image]

I also attach pcb file which was used as input:
[width9.zip](https://github.com/yaqwsx/PcbDraw/files/10834471/width9.zip)

yaqwsx, 2023-02-26:
This is how Google-chrome renders it: 

[image]

Could you try it once more with KiCAD 7? KiCAD 6 created SVG in KiCAD's native units (nm), and the browsers were confused by the huge numbers. KiCAD 7 instead uses mm, which should work everywhere.

adamws, 2023-02-27:
KiCad 7 with PcbDraw 1.1.1 works as expected :+1:

yaqwsx, 2023-02-27:
There isn't an easy fix for v6 and in the future, I would like to focus only on v7. Hence, closing.

closed 2023-02-27 by yaqwsx