yaqwsx/PcbDraw #141: Resistor color issues (various)

set-soft, 2023-03-20:
I'm trying to make my 3D resistor models generator with PcbDraw components, I'm using the following test case: 
[resistor_tht.zip](https://github.com/yaqwsx/PcbDraw/files/11020982/resistor_tht.zip)

I found various issues:
- Values under 10 ohms are incorrectly handled (0.1 and 0.01 multipliers)
- The code isn't protected against values under 0.01 ohms (I think they can't be represented, I didn't check the max values because the test case uses big resistor bodies)
- 20% tolerance isn't handled
- The code assumes values with tolerances under 5% can be represented using 4 bands. This isn't common practice. They use 5 bands, you can't represent the 1% series using 4 bands.
- Big resistor bodies aren't implemented, I think all the KiCad components should be supported.

I also disagree with the idea of having the tolerance inside the resistor value. I think the code should get it from a field named *tol* and/or *tolerance*

After various patches I got the following:

[image]

I'm trying to make it convergent with the following:

[image]

Electro707, 2023-12-27:
hello,

i'll assign myself to this, as I've created the resistor color code feature.

1. For values <10ohm, I can add implementation for that
2. Yes there is nothing in the code for values <0.01%. If a value cannot be found (such as a small multiplier), then it will result in a blank resistor
3. Added to my todo
4. Initially I just implemented a 4 color band resistor as that is the "hobbyist standard", and didn't bother adding 5 color band support. I agree that a 5-color one would be the only way to represent <1% resistors for the majority of them.
    - For a good compromise, I think pcbdraw will need to decided what color band to use (4 or 5) based off the tolerance (5 band for <=1%, 4 otherwise). When I implement this, I will also add a user override option to manually select the color bands.

6. At the time I decided to to implement the large resistors bodies, as from what I have seen they tend to be either squares or metalic bodies.

Electro707, 2023-12-29:
So I have a preliminary version of the 5-band implementation working. It selects a 5 band if the tolerance is <=2%, otherwise it sticks with a 4 band
[image]

It uses the `tol` or `tolerance` field for tolerance before it attempts to use the value field, which required more code changes on the functions that call the resistor function as the properties of the footprint needed to be passed up as an extra argument

If you want, feel free to try this feature on my branch for this: https://github.com/Electro707/PcbDraw/tree/feature_res_bands
you will also need the library with the changed resistor: https://github.com/Electro707/PcbDraw-Lib/tree/tht_res_bands

Electro707, 2024-01-01:
Update, I re-enabled the generation of the larger size resistors, after checking that KiCAD's own 3d model has the axias "bean" style for them as well.
I also added zero-ohm handling
[image]

The only symbols not created are the vertical axial ones. I might make a symbol for it, but it won't of course have the color code on it

Electro707, 2024-01-01:
[image]

I will make a PR for the symbols and this tool soon.

The math is not perfect: I already see one or two wrong resistor color codes. Planned for fixing