yaqwsx/PcbDraw #143: ModuleNotFoundError when using with pinion generate rendered

agners, 2023-04-06:
Using PcbDraw with `pinion generate rendered --renderer normal --projection orthographic` leads to the following stack trace:

```
Traceback (most recent call last):
  File "/usr/bin/pinion", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pinion/ui.py", line 72, in wrapper
    return func(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/pinion/ui.py", line 142, in generateRendered
    generate(specification=yaml.load(specification),
  File "/usr/lib/python3.10/site-packages/pinion/generate.py", line 329, in generate
    fSource, bSource = imageGenerator(board, outputdir)
  File "/usr/lib/python3.10/site-packages/pinion/ui.py", line 135, in generateImages
    return generateRenderedImages(board, outputdir,
  File "/usr/lib/python3.10/site-packages/pinion/generate.py", line 272, in generateRenderedImages
    from pcbdraw.renderer import (RenderAction, renderBoard, Side, postProcessCrop,
  File "/usr/lib/python3.10/site-packages/pcbdraw/renderer.py", line 27, in <module>
    from .pcbnew_common import findBoardBoundingBox
  File "/usr/lib/python3.10/site-packages/pcbdraw/pcbnew_common.py", line 2, in <module>
    from pcbnewTransition.pcbnew import BOX2I # type: ignore
ModuleNotFoundError: No module named 'pcbnewTransition.pcbnew'
```

I've installed PcbDraw, pcbnewTransition and pinion from source using `pip install .` . I am running Arch Linux with KiCad 7.0.1.

agners, 2023-04-06:
This change seems to fix the problem, but I am not sure if that is the proper fix:

```
--- a/pcbdraw/pcbnew_common.py
+++ b/pcbdraw/pcbnew_common.py
@@ -1,5 +1,5 @@
 from pcbnewTransition import pcbnew, isV6 # type: ignore
-from pcbnewTransition.pcbnew import BOX2I # type: ignore
+from pcbnew import BOX2I # type: ignore
 from itertools import chain
 from typing import Optional, List
 import wx # type: ignore
```

yaqwsx, 2023-11-05:
Hi, sorry for missing your issue! It seem that somehow the dependency `pcbnewTransition` that monkeypatches the KiCADs `pcbnew` module to have the same API no matter what version of KiCAD we use is not installed in your case. When you install it, everything should work.

closed 2023-11-05 by yaqwsx