yaqwsx/PcbDraw #144: Footprints are rotated

n-i-x, 2023-05-20:
Full disclosure: I'm working off a KiCad PCB that was converted to KiCad format from EasyEDA JSON. This might be the cause of issues.

Some footprints are rotated when they are plotted. I'm not sure what controls the orientation of the footprint so I don't know what to do to fix it.

[image]

I'm still working through adding footprints of the missing components and will submit PRs to the lib repo when I have them.

yaqwsx, 2023-05-21:
There is no standard or rule for footprint orientation. The 0 orientation is the orientation in which the footprint is drawn. EasyEDA might use different conventions for footprint default rotation. 

The "clean" way of resolving this is to build a library of EasyEDA footprints for PcbDraw. Any other way is non-systematic, and I would be hesitant to support it.

n-i-x, 2023-05-21:
I created a PR in #145 to allow components to be rotated without having to create a new component. This allows the standard footprint library to be mapped to an easyeda library via symlinks instead of creating rotated svgs for every footprint.