yaqwsx/PcbDraw #147: mistune 3.x - import error mistune.plugins.table

choas, 2023-06-18:
I've installed PcbDraw and get the following error
`ImportError: cannot import name 'plugin_table' from 'mistune.plugins.table' (/usr/local/lib/python3.10/dist-packages/mistune/plugins/table.py)`

The dependency installs mistune 3.0.1.
After installing mistune==2.0.5 the problem is solved.

nopeppermint, 2023-06-21:
I had exactly the same issue, as well installing mistune==2.0.5 solves the problem

set-soft, 2023-06-21:
May be a little bit of topic, but: @choas and @nopeppermint did you complain to the mistune author/maintainer?

I see the Python (Ruby, JavaScript, etc.) communities are a bunch of lazy people that doesn't dedicate some time to keep compatibility. They put a burden on the rest of people and waste time, disk space and energy in a way that I can't understand. I can understand you may want to enhance your lib and do a major release, but breaking API without providing compatibility is a bad attitude. 

And in this case thing are really bad. The v3 tag has 2 weeks and the v2 tag is from February 7 **2023**. This "guy" is doing 2 incompatible releases in less than 6 months! WTF?!

This is the kind of things that makes Python scripts a pain in the ass, you have to install the same thing many times in a system. Usually at user level with tons of ridiculous "virtual envs". This looks like the "Windows" style of solving things, much different from the "Unix" style. Some UNIX components has compatible command line flags to keep them compatible with versions created 40 years ago, not 4 months!

choas, 2023-06-21:
just as a note: mistune 2.0.0 was released on 05.December.2021

yaqwsx, 2023-06-26:
@set-soft Well, this is my fault – I wrongly specified the dependency. I should have fixed it to a major revision. The authors of mistune correctly released a major revision; unfortunately, there are numerous Python packages (e.g., hypothesis) that introduce breaking changes even for patch revision.

closed 2023-06-26 by yaqwsx

smkent, 2024-03-24:
https://github.com/yaqwsx/PcbDraw/pull/166 should resolve this.