yaqwsx/PcbDraw #148: crash when trying to create images

nopeppermint, 2023-06-21:
I tried to create some images of one of my kicad projects but it failed:

Any idea why?

```
C:\00_Git\PCB\DC_DC_and_Power_Switch_Test\V1.0>pcbdraw plot DC_DC_and_Power_Switch_Test.kicad_pcb DC_DC_and_Power_Switch_Test.jpg
Traceback (most recent call last):
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\7.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\Scripts\pcbdraw.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\ui.py", line 180, in plot
    image = plotter.plot()
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 1054, in plot
    plotter.render(self)
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 640, in render
    self._plotter.execute_plot_plan(to_plot)
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 1226, in execute_plot_plan
    action.action(action.name, os.path.join(tmp, svg_file))
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 691, in _process_baselayer
    get_board_polygon(
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 418, in get_board_polygon
    elements.append(SvgPathItem(svg_element.attrib["d"]))
  File "C:\Users\StefanWeber\Documents\KiCad\7.0\3rdparty\Python39\site-packages\pcbdraw\plot.py", line 106, in __init__
    raise SyntaxError("Unsupported path element " + path_elems[0])
SyntaxError: Unsupported path element 211.90000
```

```
Application: KiCad x64 on x64

Version: 7.0.5-28-g441c2b50b9, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: May 30 2023 05:14:45
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

nopeppermint, 2023-06-21:
seems to be the same issue as https://github.com/yaqwsx/PcbDraw/issues/142

would it be possible to generate a new release?

nopeppermint, 2023-06-21:
replacing the rectangle used for the edge cuts by 4 lines solved the issue.
But I would still prefer a new release with the fix ;-)

yaqwsx, 2023-11-05:
Duplicate #142

closed 2023-11-05 by yaqwsx