yaqwsx/PcbDraw #151: [Feature request] get ride of whitespace on generated svg images

nopeppermint, 2023-06-22:
It would be nice if svg images generated with pcbdraw would automatically be limited to whitespace.

At the moment it do this with [inkscape](https://graphicdesign.stackexchange.com/questions/21635/getting-rid-of-whitespace-in-inkscape)

set-soft, 2023-06-22:
Hi @nopeppermint ! 
Did you try `--margin`. If you are using KiBot: try with `size_detection` and you can also play with `margin`

nopeppermint, 2023-06-26:
> Hi @nopeppermint ! Did you try `--margin`

yep, does not fully solve the issue (margin 0 only applied for one side of the picture):

without --margin:

[image]

with --margin 0
[image]

nopeppermint, 2023-06-26:
inkscape -> File -> document properties -> resize to content creates then this:

[image]

set-soft, 2023-06-26:
Don't remember about the CLI interface, I no longer use it, but from KiBot you can:

1. You can specify individual margins for each side
2. Margins can be negative, removing extra space
3. You have more than one algorithm to determine the automatic margins

yaqwsx, 2023-06-26:
@nopeppermint: margin should work. If it doesn't, it is a bug. Please include the source board and provide us with the precise steps to replicate.

nopeppermint, 2023-06-26:
> @nopeppermint: margin should work.

I can confirm, it works with the Kicad Demo Project "Stickhub" but not with my Projects.
I will prepare a source board and steps to replicate

nopeppermint, 2023-06-26:
[pcbdraw_example.zip](https://github.com/yaqwsx/PcbDraw/files/11868343/pcbdraw_example.zip)

I am using Kicad:

```
Application: KiCad x64 on x64

Version: 7.0.6-rc1-31-g7bcd3e2262, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jun 24 2023 05:13:33
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

To reproduce:
- open Kicad 7.0 Command prompt
- unzip this project
- navigate to it in the Kicad command prompt
- ```pcbdraw plot pcbdraw_example.kicad_pcb default.svg```
- ```pcbdraw plot --margin 0 pcbdraw_example.kicad_pcb default.svg```
- open the generated file in inkscape to compare:

[image]
[image]

nopeppermint, 2023-06-26:
the png export seems to work without the need of --marking 0

[image]

yaqwsx, 2023-06-26:
You have silkscreen outside the PCB edges (the dimensions). This silkscreen is taken into account when apply margin. However, it is hidden by PCB substrate mask. The cropping procedure doesn't interpret the masking (as it is quite challenging to implement properly).

I will think what we can do about this, but I suggest not placing dimensions into the silkscreen layer and use, e.g., `F.Fab` instead.

yaqwsx, 2023-06-26:
> the png export seems to work without the need of --marking 0
> 
> [image]

It behaves the same as SVG export:

[image]

nopeppermint, 2023-06-26:
> It behaves the same as SVG export:

indeed, I just double checked in gimp :-(

yaqwsx, 2023-06-26:
It's exactly the same problem - we consider all geometry, but we ignore the masking at the moment.

set-soft, 2023-06-26:
> pcbdraw_example.zip

Using KiBot config:

```yaml
kiplot:
  version: 1

outputs:
  - name: PcbDraw
    comment: "PcbDraw test top"
    type: pcbdraw
    dir: PcbDraw
    options: &pcb_draw_ops
      format: svg
      warnings: none
      margin: -0.1

  - name: PcbDraw2
    comment: "PcbDraw test bottom"
    type: pcbdraw
    dir: PcbDraw
    options:
      <<: *pcb_draw_ops
      bottom: True
```

I get:
[image]
[image]

The small 0,1 mm margin is removed.
I think the problem with the standalone PcbDraw is that its trying to be "smart" and use a very complicated mechanism that involves SVG Path. But just asking KiCad which is the Bounding Box solves the problem. KiBot defaults to the KiCad mechanism, but you can ask for the SVG Path thing. KiCad knows how to clip the castellated pads.

yaqwsx, 2023-06-26:
The reason why we do it this way is that we don't want to crop components that might stick out of the PCB outline. We primarily use PcbDraw for drawing boards **with** components, not bare-bone PCBs.

The suitable algorithm might be to collect the PCB boundary from KiCAD, collect bounding boxes of all components from SVG, and combine them.

nopeppermint, 2023-06-26:
>  drawing boards with components, not bare-bone PCBs.

yes, but as far as I understand I would have to generated for each footprint an svg file as I use non of the footprints in the kicad library, all are custom..

If I run:

```
pcbdraw plot --margin 0 --no-components pcbdraw_example.kicad_pcb default.svg
```
there are no components which can be crop (as there are no-components) but the result is still the same

nopeppermint, 2023-06-26:
> 

> 

> Using KiBot config:
>
> ```yaml
> kiplot:
>   version: 1
>
> outputs:
>   - name: PcbDraw
>     comment: "PcbDraw test top"
>     type: pcbdraw
>     dir: PcbDraw
>     options: &pcb_draw_ops
>       format: svg [...]

pcbdraw plot does not accept a margin of -0.1:

```
pcbdraw plot --margin -0.1 pcbdraw_example.kicad_pcb default.svg
Usage: pcbdraw plot [OPTIONS] INPUT OUTPUT
Try 'pcbdraw plot --help' for help.

Error: Invalid value for '--margin': '-0.1' is not a valid integer.
```

yaqwsx, 2023-06-26:
> there are no components which can be crop (as there are no-components) but the result is still the same

Urg, I am just trying to explain how the current algorithm works, and why it was designed in such a way. Yes, there are limitations you run into (and we didn't consider them back then as an issue). It needs to be reworked (just like I proposed above). It will be done, sometimes. At the moment, I cannot promise you any particular timeline for the implementation.

> yes, but as far as I understand I would have to generated for each footprint an svg file as I use non of the footprints in the kicad library, all are custom..

That's kind of the point of PcbDraw - to create stylized drawings of the boards that are nice. This is why we use the hand-draw library of components - to keep them stylized. If you just want a faithful representation of the board with components, I suggest using `pcbdraw render` that just captures the 3D-rendered board and uses 3D models. Also, KiBot provides a pipeline to export 3D model and render it via Blender, IIRC. That might be also a viable path for you.

nopeppermint, 2023-06-26:
>  It will be done, sometimes. At the moment, I cannot promise you any particular timeline for the implementation.

No worries, I didn't wanted to attack you, I will change it back to a "Feature Request" and thanks for the explanations.

> pcbdraw render that just captures the 3D-rendered board and uses 3D models.

I know, but I am on windows -> not supported..

I will try Kibot.

yaqwsx, 2023-06-26:
> I know, but I am on windows -> not supported..
> 
> I will try Kibot.

Well, KiBot has the same problem. Nevertheless, both tools are designed for automated processing in CI, where Linux shouldn't be a problem. If you want to just get a picture of the PCB, why not just render it manually inside KiCAD's 3D view and save the picture?

yaqwsx, 2023-11-05:
I made changes to the algorithm for building bounding boxes in [0a137fe](https://github.com/yaqwsx/PcbDraw/commit/0a137fe1d82eefb7e8c1ceacd642d14f13fcde44). It should resolve all issue you observe.

closed 2023-11-05 by yaqwsx