yaqwsx/PcbDraw #154: Fails with KiCad 7: `TypeError: must be real number, not EDA_ANGLE`

CRImier, 2023-09-19:
```arya@lappy:~/pcb/KiCad/Reverse-engineering/usbc_debug_finder$ pcbdraw usbc_debug_finder.kicad_pcb out.svg
Traceback (most recent call last):
  File "/home/arya/.local/bin/pcbdraw", line 8, in <module>
    sys.exit(main())
  File "/home/arya/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 938, in main
    walk_components(board, args.back, lambda lib, name, val, ref, pos:
  File "/home/arya/.local/lib/python3.8/site-packages/pcbdraw/pcbdraw.py", line 515, in walk_components
    orient = math.radians(module.GetOrientation() / 10)
TypeError: must be real number, not EDA_ANGLE```

yaqwsx, 2023-09-19:
I guess you use more than a year-old version. Please upgrade before reporting further issues.

CRImier, 2023-09-19:
oh I installed it ten minutes ago using `pip install PcbDraw`! Now reinstalled using the git master version, works, thank you. ig the Kicad7 changes aren't yet in the main release.

also, thank you for the prompt response!

closed 2023-09-19 by CRImier

yaqwsx, 2023-09-19:
That's strange; the latest version [v1.1.2](https://github.com/yaqwsx/PcbDraw/releases/tag/v1.1.2) is on Pypi: https://pypi.org/project/PcbDraw/. And clearly you had version <1 installed.